vhelin/wla-dx #500: Macro labels break childLabels

lajohnston, 2022-04-22:
Hi there,

I'm running into a strange issue when using macro and child labels. The following code breaks on the commented line:

```
.macro "myMacro"
    jp nz, \@macroLabel
    \@macroLabel:
.endm

.section "mySection" force
    mySection:
        jp @childLabel  ; FIX_REFERENCES: Reference to an unknown label "mySection@childLabel"
        myMacro         ; Removing this or moving it up one line fixes the error above

        @childLabel:
.ends
```

It works if I remove the \@macroLabel label in the macro and replace it with a + label, but the documentation warns against using unnamed labels in macros. It also works if I move the macro call up one line, so it's as if calling myMacro changes the context and doesn't restore it afterwards, so I can't reference `@childLabel` after that point. I can remove the child label for now so it's not a blocker, but just an FYI!

This is happening with a fresh build from master, but also happened with the old version I was using so doesn't seem to be a new thing.

Thanks

vhelin, 2022-04-22:
It's been an exhausting week, but if I read this correctly your myMarco creates a new root label \@macroLabel so all @childLabels after that point to it. This is kind of how things work, if I'm correct...

So after myMacro call @childLabel: will be actually be 0macroLabel@childLabel ?

I'll recheck this issue this weekend, just in case I've missed something. :)

vhelin, 2022-04-22:
And FYI, .MACROs operate in the global context, a .MACRO call is just a jump for the parser, jump to the beginning of the .MACRO, lexically.

lajohnston, 2022-04-24:
Ah that makes sense now thanks. I've only just started using child labels but I think I'll stick to private labels using the _ prefix as they seem more robust in this situation. I've been running into a few hard to debug problems with labels in macros recently but I think they all stem from the same sort of thing. It wasn't until recently I realised it's unwise to use anonymous + ++ labels inside macros as they can get mixed up with other anonymous labels in the context the macro is called.

Are there any thoughts or ideas around making macros labels work within their own context (or at least, just the _private ones and anonymous ones)? It seems any label defined in a macro kind of needs both a \. for the macro name and \@ for the call count to provide uniqueness, so I almost wish this was the default unless there's a use-case I don't know of where you wouldn't want this. If anonymous labels had a unique context identifier added to them they could also avoid getting muddled up.

vhelin, 2022-04-24:
I think a .SECTION is the context for anonymous labels? But yeah, currently you need to manually construct unique labels inside .MACROS... 

I'm thinking that I should add something like KEEPLABELSLOCAL etc. to .MACRO that would return the old root label when .ENDM is parsed to it wouldn't affect child labels outside .MACRO... ?

lajohnston, 2022-04-25:
yes that's fair. Have you considered using semver for the version numbers so you could then reserve any breaking changes for major releases? That said I don't think this issue is particularly high priority if no one else has reported it until now.

My dream would be a keyword that makes all labels in a macro have some uniqueness to them so they exist in their own context, almost like the namespacing feature, i.e.:

```
.macro "foo" namespaced args bar baz
    myLabel:    ; becomes foo0.myLabel
    jp +        ; jumps to foo0.+
    +:          ; becomes foo0.+
.endm           ; restores root label
```

I don't know how the magic works behind + ++ +++ that allows them to be defined multiple times but it would be quite handy if it worked here too.

vhelin, 2022-05-02:
> yes that's fair. Have you considered using semver for the version numbers so you could then reserve any breaking changes for major releases? That said I don't think this issue is particularly high priority if no one else has reported it until now.

We just released "recently" v10.0 that broke a couple of things, not thinking of release v11.0 any time soon, and not really going to change the default behavior of .MACROs... :)

lajohnston, 2022-05-04:
> not really going to change the default behavior of .MACROs... :)

no, the suggestion I gave above would be an opt-in feature by passing a parameter to `.macro`. This would be similar to your suggestion to restore the context with `KEEPLABELSLOCAL`, but would go one step further and keep all the labels defined in the macro within their own context (either with namespacing or some other mechanism). I used the word `namespaced` as the parameter, but that's just a placeholder:

> ```
> .macro "foo" namespaced args bar baz
>     myLabel:    ; becomes foo0.myLabel
>     jp +        ; jumps to foo0.+
>     +:          ; becomes foo0.+
> .endm           ; restores root label
> ```

I think the minimum to close this GitHub issue would be to just add a note regarding child labels in the macro 'CAVEAT' section of the labels documentation, but I'd love to see some quality of life improvements for macro labels at some point. As it stands, writing code in macros is pretty clunky compared to writing code in sections, especially with the absence of reusable un-named labels:

```
.macro "myMacro"
    \.\@someLabel:
        jp z, \.\@zero
        \.\@zero:
    
        jp z, \.\@zero2
        \.\@zero2:
.endm
```

Compared to:
```
.section "mySection" free
   someLabel:
        jp z. +
        +:
      
        jp z. +
        +:
.ends
```

vhelin, 2022-05-04:
"... with the absence of reusable un-named labels"

Not sure I understand you. Don't "+" work also inside a .MACRO? Do the latest commits solve this (keyword ISOLATED)?

vhelin, 2022-05-04:
"I think the minimum to close this GitHub issue would be to just add a note regarding child labels in the macro 'CAVEAT' section of the labels documentation..."

doc/asmsyntax.rst ("Labels" section) mentions this issue and how to use the keyword ISOLATED to solve the problem. Also doc/asmdiv.rst has an example about this and the keyword ISOLATED.

lajohnston, 2022-05-04:
Great thanks! 🎉  The new ISOLATED keyword does the trick :) I couldn't reproduce the un-named label issue but I have been bitten by it in the past. I've just tried with latest `master` with the example and it's working fine.

One thing to clarify, the doc says:
> To make local labels inside a .MACRO isolated, and the previous example to work, use the keyword ISOLATED

My understanding of local labels is that they are ones prefixed with an underscore and are local to the section. The above though implies you can use those inside a `.macro` and they will be isolated to that macro. I've just tried this out and it seems that they're isolated to the `.section` that calls the macro and not the macro itself, whether you use ISOLATED or not.

Do you have plans to isolate them to the macro or is using `\.\@` still recommended for named macro labels? If you aren't, the doc might be clearer if it read:
> To make un-named labels inside a .MACRO isolated, and the previous example to work, use the keyword ISOLATED

vhelin, 2022-05-04:
"To make local labels inside..." I think I should have written "To make un-named labels inside..." there. Gotta hit the bed, 00:27 here, but I'll continue tomorrow...

Adding that .MACRO context to local labels could also be done, and pretty easily, I think. Just thinking that should we have keyword EXTRAISOLATED for that?

lajohnston, 2022-05-05:
> Adding that .MACRO context to local labels could also be done, and pretty easily, I think.

That would be great if possible! :)

> Just thinking that should we have keyword EXTRAISOLATED for that?

If anything I think ISOLATED implies it should isolate everything already. Later on you could always add more granular options like ISOLATEUNNAMED and ISOLATELOCAL if someone presented a good use case for needing them.

vhelin, 2022-05-05:
> If anything I think ISOLATED implies it should isolate everything already. Later on you could always add more granular options like ISOLATEUNNAMED and ISOLATELOCAL if someone presented a good use case for needing them.

Thanks for the idea, did just that :)

lajohnston, 2022-05-05:
These are all great thanks! They make labels in macros much cleaner and more reliable so I'll be using them for all my macros going forward. They fix the original issue I was having with the child labels so I'm happy to close this :) Thanks again!

closed 2022-05-05 by lajohnston