vhelin/wla-dx #501: SECTION attribute to prevent crossing a page boundary

Kroc, 2022-04-30:
This is kind of like alignment, but the ability to specify that a `SECTION`/`RAMSECTION` must *not* cross an 8-bit boundary.
If a section can be guaranteed not to cross into a 9th bit, then I can make Z80/SM83 code faster by only incrementing an 8-bit counter instead of a 16-bit counter. I just got bitten by this today when code I added bumped a table over a boundary and my code exploded unintentionally!

This is not quite the same as `ALIGN`, as the actual position with the 256-byte range may not be important at all, and even with `ALIGN` you still may not want a section to go over the 8-bit boundary.

I don't really have a suggestion what keyword could be used for this.

vhelin, 2022-04-30:
Have you checked out WINDOW and BITWINDOW? .SECTION has those and I think they should do the job...

Kroc, 2022-04-30:
These aren't in the current online documentation and the documentation isn't included in releases either :/

vhelin, 2022-04-30:
Yes they are in the documentation in v10.1 release. Yes they are here: https://www.villehelin.com/wla-README.html

No, they are not in the Documentation linked in README.md, for some reason. Which is kind an issue... Why https://wla-dx.readthedocs.io/en/latest/ is out-of-date? This needs to be solved...

vhelin, 2022-04-30:
Ok, Read the Docs can now build the latest documentations.

closed 2022-04-30 by vhelin

Kroc, 2022-04-30:
Ah, `BITWINDOW` is perfect, thanks!