vhelin/wla-dx #505: Create a .bat version of run_tests.sh for Windows CMD

vhelin, 2022-05-15:
Currently the Windows MSVC Azure Pipelines script doesn't run the tests as we don't have a .bat version of the run_tests.sh script...

cr1901, 2022-05-17:
This in principle should've been 30 minutes for me to add for you, but I've had computer trouble this past week and had to reinstall Windows. I'll see what I can do for getting this done...

vhelin, 2022-05-17:
Nice! I practically know nothing about .bat scripting...

cr1901, 2022-06-19:
My opinion after spending a bit of time implementing this is that "you are better off installing `sh`, `make`, `rm`, and friends, and just running `sh run_tests.sh`, even from Windows. This works for me locally, anyway.

vhelin, 2022-06-19:
Ouch. 

> My opinion after spending a bit of time implementing this is that "you are better off installing sh, make, rm, and friends, and just running sh run_tests.sh, even from Windows. This works for me locally, anyway.

Then the question/issue turns into this: How to install those tools in Azure Pipelines / Windows MSVC script?

cr1901, 2022-06-24:
Btw, I forgot to ask... is `bsh` (`run_tests.bsh`) on Amiga an older version of `bash` or a different shell? For Amiga OS 3.x or 4.x?

vhelin, 2022-06-24:
If I interpret bsh.doc correctly, the Amiga bsh I'm using is v0.98. Feels pretty simple and old. Amiga OS 3.1.

[image]

vhelin, 2022-06-24:
> Ahhh so, even Amiga OS 3.x used rm for remove, and won't error if file
> doesn't exist if -f was provided.
>
> Does Azure pipelines come with make? If not maybe instead of using the
> Makefiles, I emulate the behavior inside the PowerShell script via
> functions that run the commands sequentially.

Amiga uses "delete" for remove, but bsh seems to provide "rm". Also running
"bsh run_tests.bsh" on Amiga doesn't error/break in any case so I'll need
to look at the output (for the 15 minutesish it runs) to spot any errors.
Fortunately console's update rate is slow when it scrolls...

Azure Pipelines / Windows could not run run_tests.sh last time I tried,
don't remember what it was missing. Most probably make etc...

vhelin, 2022-07-08:
I was just messing around the Azure Pipelines .yml files and noticed that when we build Windows exes using Visual Studio solution file we also run the tests (run_tests.sh -windows-x86)..

> Currently the Windows MSVC Azure Pipelines script doesn't run the tests as we don't have a .bat version of the run_tests.sh script...

... so my initial issue didn't actually exist. :D Me and my (bad) chicken memory...

closed 2022-07-08 by vhelin