vhelin/wla-dx #508: Automatic PRG address is wrong

Kroc, 2022-05-24:
At some point, the automatic PRG address has started giving the wrong value. I rely on this so that the PRG starting address does not have to be hard-coded into the build scripts.

Given a memory map as such:
```
.MEMORYMAP
        ;-----------------------------------------------------------------------
        DEFAULTSLOT     2
        SLOTSIZE        $0100
        SLOT 0          $0000   "ZP"
        SLOT 1          $0100   "STACK"
        
        ; main program RAM
        ; ("ROM" in WLA-DX parlance)
        ;
        SLOTSIZE        $FDFA   ; $FFFA-$FFFF are 6502 interrupt vectors
        SLOT 2          $0200   "MAIN"

        SLOTSIZE        $1000
        SLOT 3          $C000   "HIRAM"
        SLOT 4          $D000   "IO"
        SLOTSIZE        $2000
        SLOT 5          $E000   "KERNAL"
.ENDME
```

And the Commodore BASIC bootstrap like this:

```
.SLOT   "MAIN"
.ORGA   $0801

;===============================================================================
cbmbasic:

        .WORD @end              ; pointer to next BASIC line

        .WORD 64                ; line-number
        .BYTE $9e               ; `SYS`

        ; address of machine-code address in decimal:
        .BYTE '0' + @asm # 10000 / 1000
        .BYTE '0' + @asm #  1000 /  100
        .BYTE '0' + @asm #   100 /   10
        .BYTE '0' + @asm #    10
        
        .BYTE 0                 ; end-of-line

@end:   .WORD $0000             ; end of BASIC program
@asm:
```

WLA-DX used to select `$0801` (as expected) for the PRG file, but now uses `$0601`, probably since 10.0; I haven't been able to build older versions just yet due to moving to VS2022.

vhelin, 2022-05-24:
-> create a symbol file and check out what label is at $601...

vhelin, 2022-06-02:
Did you manage to find out which label was at $0601 in your project?

Kroc, 2022-06-02:
Sorry, I need to get back to you on this one, going through a rough patch at the moment and I need to get builds working with VS2022

Kroc, 2022-06-03:
I managed to cleanly compile on VS2022. Firstly, "wla6510.exe" will no longer link (wrong version number), so I switched to "wla6502.exe". This is the output for https://github.com/Kroc/zsheet:

Assembler:
```
Pass 1...
Directive checks...
Internal pass 1... 
Internal pass 2...
-------------------------------------------------
---                   ROM                     ---
-------------------------------------------------
ROM bank 0 (1069 bytes (1.64%) used)
  - Outside .SECTIONs (731 bytes)
    - Used space at $0000-$02da (731 bytes).     
  - Sections (338 bytes)
    - .SECTION "fast_write" (338 bytes).
-------------------------------------------------
---                   RAM                     ---
-------------------------------------------------
RAM slot 0 (ZP) bank 0 (76 bytes (29.69%) used)  
  - .RAMSECTION "ZP_00" (2 bytes).
  - .RAMSECTION "ZP_90" (2 bytes).
  - .RAMSECTION "ZP_93" (3 bytes).
  - .RAMSECTION "ZP_97" (4 bytes).
  - .RAMSECTION "ZP_9D" (1 bytes).
  - .RAMSECTION "ZP_A0" (5 bytes).
  - .RAMSECTION "ZP_AE" (2 bytes).
  - .RAMSECTION "ZP_B7" (6 bytes).
  - .RAMSECTION "ZP_C1" (7 bytes).
  - .RAMSECTION "ZP_CB" (14 bytes).
  - .RAMSECTION "ZP_F3" (4 bytes).
  - .RAMSECTION "ZP_REG" (3 bytes).
  - .RAMSECTION "ZP_TEMP" (4 bytes).
  - .RAMSECTION "ZP_SCREEN" (6 bytes).
  - .RAMSECTION "ZP_COPY" (4 bytes).
  - .RAMSECTION "zp_drawrc" (3 bytes).
  - .RAMSECTION "ZP_CURSOR" (6 bytes).
RAM slot 2 (LORAM) bank 0 (14 bytes (6.86%) used)
  - .RAMSECTION "ZS_WORK" (14 bytes).
-------------------------------------------------
---                 SUMMARY                   ---
-------------------------------------------------
ROM: 1069 bytes (1.64%) used.
RAM: 90 bytes used.
```
Linker:
```
Using the address $0 (of label "RAM_USAGE_SLOT_ZP_BANK_0_START") as the load address for the PRG.
-------------------------------------------------
---                   PRG                     ---
-------------------------------------------------     
PRG $0000-$0451 (37 bytes (3.35%) free)
  - Free space at $02db-$02ff (37 bytes)
-------------------------------------------------     
---                   RAM                     ---     
-------------------------------------------------     
RAM slot 0 (ZP) bank 0 (180 bytes (70.31%) free)      
  - Free space at $001c-$008f (116 bytes)
  - Free space at $0092-$0092 (1 bytes)
  - Free space at $0096-$0096 (1 bytes)
  - Free space at $009b-$009c (2 bytes)
  - Free space at $009e-$009f (2 bytes)
  - Free space at $00a5-$00ad (9 bytes)
  - Free space at $00b0-$00b6 (7 bytes)
  - Free space at $00bd-$00c0 (4 bytes)
  - Free space at $00c8-$00ca (3 bytes)
  - Free space at $00d9-$00f2 (26 bytes)
  - Free space at $00f7-$00ff (9 bytes)
RAM slot 2 (LORAM) bank 0 (190 bytes (93.14%) free)   
  - Free space at $000e-$00cb (190 bytes)
-------------------------------------------------
---           HEADERS AND FOOTERS             ---     
-------------------------------------------------     
2 additional bytes from the CBM PRG header.
-------------------------------------------------     
---                 SUMMARY                   ---     
-------------------------------------------------     
PRG: 37 bytes (3.35%) free of total 1106.
RAM: 370 bytes (80.43%) free of total 460.
```

vhelin, 2022-06-04:
Thanks! It seems that RAM_USAGE_SLOT_ZP_BANK_0_START is a label that WLALINK has generated itself, so now I made WLALINK to skip those generated RAM_USAGE_SLOT_* labels when trying to find that PRG address automatically. If you try the latest sources (I couldn't build your project using build.bat, it gave me an "Access denied" error), does it work better?

Kroc, 2022-06-04:
Fixed with recent patch, thank you!

closed 2022-06-04 by Kroc