vhelin/wla-dx #509: Integer Divide

Kroc, 2022-06-04:
I've just run into a problem that requires an integer divide. I'm trying calculate a table of divisions and remainders, but there's not a way to do an integer divide to give the remainder!

```
copy_rows_iterations:
        ;-----------------------------------------------------------------------
        ; for a given number of rows, how many times a full-byte's worth of
        ; rows should be processed. the remainder is given in the next table
.REPEAT SYS_SCREEN_ROWS INDEX R
        .BYTE (R / (256 / SYS_SCREEN_COLS))
.ENDR

copy_rows_remainder:
        ;-----------------------------------------------------------------------
.REPEAT SYS_SCREEN_ROWS INDEX R
        .BYTE ((R+1) * SYS_SCREEN_COLS) # ((256 / SYS_SCREEN_COLS) * SYS_SCREEN_COLS)
.ENDR
```

Here, `(256 / SYS_SCREEN_COLS)` returns a float, 6.5 for a screen width of 40 cols, rather than 6.0. This means that the result of the calculation on the right-hand-side of the modulo is not 240.

In this instance I need either an integer division operator, or some form of `round` | `floor` | `ceil` function.

vhelin, 2022-06-04:
I think I'll add round(), floor() and ceil() tomorrow, should be an easy job...

Kroc, 2022-06-05:
Fixed, thanks!

closed 2022-06-05 by Kroc