vhelin/wla-dx #510: WDM instruction unimplemented

jeffythedragonslayer, 2022-06-07:
Super Nintendo developers use `WDM 0` to insert breakpoints, but WLA doesn't understand it:

```
E:\bazz\VRAM-SNES-Program>wla-65816 -o vram.obj vram.asm
vram.asm:90: ERROR: Unknown symbol "WDM".
```

vhelin, 2022-06-08:
It's defined in opcodes_65816.c in the official WLA.

[image]

I just made a test and it assembles correctly (-> $42).  Are you using Bazz's version of WLA DX?

vhelin, 2022-06-08:
It seems that WDM takes another byte and WLA is currently wrong. I'll fix it...

vhelin, 2022-06-08:
Ok, do the latest sources fix the issue (they should)?

jeffythedragonslayer, 2022-06-09:
I came across this bug while following one of bazz's tutorials that is why his name is in my path

closed 2022-06-10 by jeffythedragonslayer