vhelin/wla-dx #511: confusion over what is an opcode/instruction/mnemonic

jeffythedragonslayer, 2022-06-10:
I suggest renaming the opcode struct and files; from what I hear on the 6502 forums (see http://forum.6502.org/viewtopic.php?f=12&t=6850#p88333) and https://undisbeliever.net/snesdev/65816-opcodes.html the opcode refers only to the two hex digit code.  The 3 letter abbreviation is called the mnemonic, which when "conjugated" with an addressing mode and operands (though some are so simple they don't need to be conjugated with anything,) becomes an instruction.

vhelin, 2022-06-10:
I hope what I did was enough. If it wasn't, let me know. Also volunteers are accepted for any additional renaming job. :)

jeffythedragonslayer, 2022-06-13:
This is too big of a change for me to want to review every line, but from what I skimmed over it looks good :)

closed 2022-06-13 by jeffythedragonslayer