vhelin/wla-dx #512: padding a STRUCT to a given size (or add INSTANCEOF spacing between elements?)

sverx, 2022-06-14:
I wonder if there's a way to force the size of a STRUCT to a set value. :thinking: 

Say I have a STRUCT called _object_ that has many fields in it, and that I know I'll be constantly adding/replacing/removing fields until the end of the project, but at the same time I have to make sure that the size doesn't change so that the objects elements constantly keep the same distance in the array refined in RAM:

```
.define MAX_OBJECTS    12

.ramsection "objects" slot 3
  objects instanceof object MAX_OBJECTS       ; all the objects
.ends
```

So I guess I'm either looking for a simple workaround or I'm requesting an additional feature like
`.STRUCT enemy_object PADDING 64`
for instance?

Thanks!

vhelin, 2022-06-14:
I have to say I don't remember .STRUCT's definition that well so you might be asking for a new feature.

sverx, 2022-06-14:
> Do you want a .STRUCT instance to always be the same size (X bytes) regardless of the elements inside with padding added perhaps to the end of the elements/fields to reach X bytes.

Yes, exactly, and the padding should be at the end.

It could be added either into the STRUCT definition, as in my example:
`.STRUCT object PADDING 64`
or I guess it's probably even better in your way so that it's only padded when expressly requested
`objects INSTANCEOF object 12 INSTANCESIZE 64`

Either of them would work fine :heart:

sverx, 2022-06-14:
Well of course the padding on the _object_ struct definition would also affect the value of __sizeof_object_ whereas declaring a different size for the instance in the `.ramsection` wouldn't absolutely have that effect.

So in that second case one should be careful and probably use a define such as
`.def PADDED_OBJECT_SIZE  64`
and then use it when needed so both in the declaration:
`objects INSTANCEOF object 12 INSTANCESIZE PADDED_OBJECT_SIZE`
and while iterating the array:
```
[...]
ld de,PADDED_OBJECT_SIZE
add hl,de
[...]
```

vhelin, 2022-06-14:
That's true. Perhaps it would make sense to add the size (can't really specify padding as the amount of padding is calculated so that the .STRUCT has a fixed size) to .STRUCT definition - that should lead to less problems?

```
.STRUCT object SIZE 64
```

?

vhelin, 2022-06-14:
Then again I guess I could add both SIZE to .STRUCT and INSTANCESIZE to .RAMSECTION's INSTANCEOF...

vhelin, 2022-06-14:
SIZE would be in line with SIZE in .SECTION and in .RAMSECTION as there we add padding as well...

cr1901, 2022-06-14:
>can't really specify padding as the amount of padding is calculated so that the .STRUCT has a fixed size

**This can be a problem for future-us**, but it may be useful to be able to use the assembler-calculated padding size between members (as well as the padding added to the end) in calculations if this can't already be done. This is basically a combo of how one might use `offsetof` and `sizeof` in C. But let's not add this until someone needs it.

sverx, 2022-06-14:
Well, when specifying the SIZE attribute of a struct one could have some __paddingof_object_ constant defined beside the usual __sizeof_object_ :wink: 

Also of course if the struct definition can't fit into the SIZE provided value, an error must be thrown.

vhelin, 2022-06-14:
I've started to work on this. I don't think I can fix this issue today, but perhaps tomorrow or on the day after that.

vhelin, 2022-06-14:
I'll continue working from this point next evening...

sverx, 2022-06-17:
So you're adding the COUNT keyword but you're not making that compulsory - it means that all these would work

`objects INSTANCEOF object SIZE 64 COUNT MAX_OBJECTS`

`objects INSTANCEOF object COUNT MAX_OBJECTS SIZE 64 `

`objects INSTANCEOF object SIZE 64 MAX_OBJECTS`

`objects INSTANCEOF object MAX_OBJECTS SIZE 64`

`objects INSTANCEOF object MAX_OBJECTS`

correct? If so, this is pure gold :heart_eyes:

vhelin, 2022-06-17:
SIZE must come right after INSTANCEOF so these will work

objects INSTANCEOF object SIZE 64 COUNT MAX_OBJECTS
objects INSTANCEOF object SIZE 64 MAX_OBJECTS
objects INSTANCEOF object MAX_OBJECTS

If SIZE is given after COUNT then in my head that would mean that the size of the instantiated objects, all, would be SIZE.

If you think that that's wrong I'll make it so that SIZE can be given also after the number of instances. :)

sverx, 2022-06-17:
Is there any other directive where the order of attributes matter? If it isn't so, I wouldn't add any that breaks this rule... so we should probably accept that any order is valid even when the resulting definition sounds 'wrong'.

On the other hand, if order matters in a few circumstances already, we might force SIZE to be the first attribute when present to be valid, return an error otherwise.

vhelin, 2022-06-17:
I feel like there are directives where the order still matters, but making
it so that it doesn't matter is the way to go, if you ask me. I have no
time today to work on this issue, but tomorrow I can...

sverx, 2022-06-17:
I agree that making it so that the order doesn't matter is way better.
Also, no rush, I will survive even without this :rofl: 
Thanks!

vhelin, 2022-06-20:
@sverx What do you think, is the issue now solved?

sverx, 2022-06-21:
This seems to work properly, at least using it either with

```
.struct object SIZE 64
[...]
```

or with

```
.ramsection "objects variables" slot 3
  objects instanceof object MAX_OBJECTS SIZE 64
[...]
```

closed 2022-06-21 by sverx