vhelin/wla-dx #515: GPL-2.0 is a deprecated SPDX license

chenrui333, 2022-06-24:
[GPL-2.0](https://spdx.org/licenses/GPL-2.0.html) is deprecated by SPDX, please consider using [GPL-2.0-only ](https://spdx.org/licenses/GPL-2.0-only.html) or [GPL-2.0-or-later](https://spdx.org/licenses/GPL-2.0-or-later.html). Thanks!

relates to Homebrew/homebrew-core#104357

vhelin, 2022-06-30:
See discussion https://github.com/vhelin/wla-dx/discussions/516 for more about this... I'm hoping we can decide which one we'll use in the coming weeks.

vhelin, 2022-07-09:
Ok, if there are no objections, I'm changing the license to be GPL-2.0-or-later at the end of July 2022...

vhelin, 2022-08-23:
Sorry, got (a longish) COVID19 at the end of July and totally forgot this issue... Anyway, I'll change the license to GPL-2.0-or-later this week if no objections are written here.

vhelin, 2022-08-23:
I hope that what I did was enough to change the license to GPL-2.0-or-later. If it was not, please let me know how to fix the issue.

vhelin, 2022-08-28:
No objections or improvement ideas or anything so I'm closing this issue.

closed 2022-08-28 by vhelin