vhelin/wla-dx #518: WLALINK and SPC700 instruction JMP [!?+X] (opcode $1F)

Ramsis-SNES, 2022-07-01:
Trying to assemble this SPC700 code:

`jmp   !(Table_08e9+x)`

WLA-SPC700 is fine with this (until I try and change the syntax in any way, so it seems correct), but WLALINK complains:

_PARSE_STACK: Unresolved reference to "x"._

I guess WLALINK wants to do a calculation here instead of recognizing x as the X register (replace `x` with `nothing`, and the same thing happens, with `nothing` in the error message).

Assembled instruction should be 1F E9 08.

Any help is appreciated! :-)

Project code (no make file yet, but the two commands for assembly are in the .asm file):
[spc700-driver-v1.4-reassembly.zip](https://github.com/vhelin/wla-dx/files/9031908/spc700-driver-v1.4-reassembly.zip)

vhelin, 2022-07-02:
Currently WLA-SPC700 seems to support two kinds of jumps

```
  { "JMP !?", 0x5F, 2 },
  { "JMP [!?+X]", 0x1F, 2 },
```

... so either "JMP !Table_08e9" or "JMP [!Table_08e9+X]" work. I guess you are trying to use the latter.

vhelin, 2022-07-02:
I Googled a little around the Internet, and didn't see anywhere the format you are trying to use:

```
jmp !(Table_08e9+x)
```

Where did you get that from?

```
jmp [!Table_08e9+x]
```

seems to be the correct format?

Ramsis-SNES, 2022-07-02:
Ouch, thank you so much, Ville! :+1: :+1: :+1: 

I even stared at the SPC700 instruction list and saw the square brackets, but as WLA-SPC700 stopped complaining about the syntax after I added the `!` myself, it never occurred to me that was the issue. XD Thanks again! :-) :1st_place_medal: 

> I Googled a little around the Internet, and didn't see anywhere the format you are trying to use:
> 
> ```
> jmp !(Table_08e9+x)
> ```
> 
> Where did you get that from?

The bsnes-plus debugger/tracer. As you can see, it even omits the `!` for some reason.

```..0234 jmp   ($08e9+x)         A:00 X:02 Y:00 SP:01ef YA:0000 ......Z.```

I don't even like bsnes-plus all that much, it just seems to be the most reliable piece of SNES debugging software around atm.

Sorry for bothering you, I'm rather new to SPC700 coding. Again, thanks A LOT for your input! :D

closed 2022-07-02 by Ramsis-SNES