vhelin/wla-dx #519: Make .HEX more useful/flexible

Ramsis-SNES, 2022-07-01:
.HEX currently only accepts a string in the form of `aaBB55....` as an argument.

It would be a great improvement if .HEX would support hex values directly copied from a hex editor, like `AA BB 55 .. ..`. It would be even better if the byte data didn't have to be given as a string because then line breaks could possibly be added as well. If required, have a non-string hex block be terminated with .ENDHEX.

vhelin, 2022-07-02:
How is it now?

```
``.HEX`` can also be used in the following ways::

    .HEX 01 AA 02 BB 03 CC    ; -> .DB $01, $AA, $02, $BB, $03, $CC
    .HEX BLOCK
         01 02 03 04 05 06    ; -> .DB $01, $02, $03, $04, $05, $06
         07 08 09 0A 0B 0C    ; -> .DB $07, $08, $09, $0A, $0B, $0C
    .ENDHEX
```

Ramsis-SNES, 2022-07-02:
Looks great, thanks! I'll try it out later. :D

Ramsis-SNES, 2022-07-02:
Works like a charm, thank you for a lightning-fast feature enhancement! :D

closed 2022-07-02 by Ramsis-SNES