vhelin/wla-dx #521: Issue with expressions in `[...]`

Kroc, 2022-07-05:
I have a file that's in a namespace, and using an expression, I can't refer to a local RAM label;

Works:
```
ld      [label], A
```
Does not work:
```
ld      [label+HI], A
```
`PARSE_STACK: Unresolved reference to "label".`

Works:
``
ld      [namespace.label+HI], A
```

vhelin, 2022-07-05:
I'll try to fix this later this week. I didn't implement namespaces so I have no idea how they work. I hope the code is clear and simple... :)

vhelin, 2022-07-08:
For some reason, probably an ancient decision, WLA-GB didn't release unused, pending calculations that were created while parsing the assembly code. Some of these unused calculations had label references without a namespace reference. Didn't really fix the unused, pending calculations, but made WLA-GB to free them and that at least fixed my testcase.

I do assume here that you got this issue by writing GB-Z80 code... If not, let me know. :)

Kroc, 2022-07-08:
Yes, this is GBZ80. Recompiled latest and this issue is fixed, although note that I can't build an x64 binary as it says "Unknown WLA_NAME!".

vhelin, 2022-07-08:
What OS, what build script?

Kroc, 2022-07-08:
WIN10 with VS2022 using the VS solution

vhelin, 2022-07-08:
Hmm I have VS2017 and also I cannot build x64 binaries. I'll try to solve that next...

[image]

vhelin, 2022-07-08:
Just pushed a fix to the Visual Studio solution file. If you try the latest sources x64 building should once again work...

Kroc, 2022-07-08:
Works in VS2022, thanks 👍