vhelin/wla-dx #523: Struct use seems broken by SPC700 instruction set

AmNotOnline, 2022-07-09:
In my code, I use a struct to create 4 helper variables in zeropage, for when I need pointers that don't need to sit around all the time, or for when the internal registers just aren't up to the task. In my header file, they are defined as such:

```
.STRUCT HELPERVARIABLES
  N0  DB
  N1  DB
  N2  DB
  N3  DB
.ENDST

helpervariable INSTANCEOF HELPERVARIABLES
```

And in my code, they are used as such:
```
  ASL A                       ; Entries are 2 wide
  MOV X, A
  MOV A, [returnStack + X]
  MOV helpervariable.N0, A
;     ^^^^^^^^^^^^^^^^^ Causes problems
  INC X
```
When trying to assemble, I get the error code: `CODE\INSTRUCTION_DECODE.SPASM:10: ERROR: Unknown symbol "MOV".`
Using a normal label works fine, which makes me think that it's the dot causing the issue, as the SPC700 implements it for instructions that move a specific bit from memory into the carry flag.

vhelin, 2022-07-09:
I'm looking at WLA's code and it seems that the dot handling is the culprit here, you might be correct...

vhelin, 2022-07-09:
If you try the latest sources the issue should now be fixed, I hope. If it's not, please let me know! I might redo this fix differently one day in the future, if I can find a nicer way to do it.

AmNotOnline, 2022-07-10:
Yes, the bug has been fixed. Thank you ever so much!

closed 2022-07-10 by vhelin