vhelin/wla-dx #525: Operand hint fails to properly override operand size

Scryfox, 2022-07-17:
Hello! First wanted to say, huge fan of your work on this, it is amazing. I just wanted to bring up an issue I had and check if I am being dumb of if there might be a bug. The actual code was simple enough:
```
lda    #$01.B
sta    $2105
```

But it generated the following code upon compilation:
```
lda    #$8D01
ora    $21
```
The last setting to the A & X/Y registers was `rep    #$30`.
Upon adding a `sep    #$20` ahead of it, it compiled properly.

Some background:
Compilation target is 65816, issue was observed during V-Blank so maybe that is part of it?
The disassembly came from the MESEN-S emulator if that means anything.

Thanks again for all your hard work!

Ramsis-SNES, 2022-07-17:
Hi! :3

Just chiming in because this is a known trap for young players™: If your accumulator is set to 16 bits via the status flag (`rep #$30`), then an instruction like `lda` will fetch 16 bits no matter what. The processor just interprets the next byte (`sta`) as the high byte of a 16-bit operand, hence the `lda #$8D01`; the following byte is then assumed to be the next instruction (`$05` is in fact `ora dp`).

In other words, you need to make sure to set the processor status correctly beforehand, although WLA will assemble your code as you wrote it, even if it is inconsistent (`lda #$01.b`, 8 bits) with the actual processor status (16 bits) as the actual instruction (`lda #const` or `$A9` in this case) is the same regardless.

Hope this helps to clear up any misconceptions. Happy coding! :-)

Scryfox, 2022-07-17:
I appreciate the heads up! I mainly asked as I noticed this in the documentation:
> In WLA-65816 .ACCU/.INDEX/SEP/REP override .8BIT/.16BIT/.24BIT when considering the immediate values, so be careful. Still, operand hints override all of these, so use them to be sure.

This is really what got me confused, as it suggests that the hints should override even the opcodes. I think I am going to do as you suggest regardless and be more explicit, but just wanted to point this out as it seems to be a bug regardless. Thanks!

Ramsis-SNES, 2022-07-17:
Actually, it doesn't seem to be a bug at all. :D

Consider this:

- The accumulator is set to 16 bits, i.e. the 65816 is in 16-bit mode when it hits the `lda` instruction.
- You use an 8-bit operand hint (`.B`) to override the assembler's operand size detection. So instead of assembling the instruction to `A9 01 00`, WLA does exactly as instructed and assembles it to `A9 01` instead. (Look at your assembled code in a hex editor to verify that the extra `00` is missing.)
- Here's the thing: _The processor doesn't know, and it doesn't care._ It's in 16-bit mode, right? So the processor has your `lda` instruction take two bytes as an operand, no matter what they are. Thus `A9 01 8D` becomes `lda #$8D01` in your disassembly (where `8D` is the byte immediately following an 8-bit operand, i.e. the intended `sta` instruction), and your program breaks.

The key to understanding this is the fact that

1. overriding operand sizes in WLA at any point and
2. _telling_ the processor what operand size to use by setting its status register appropriately

are two completely different things you need to take care of separately.

HTH :-)

Scryfox, 2022-07-17:
Dang, I am seeing what you are saying lol, nice. For some reason I was thinking WLA was gonna add a SEP/REP instruction to address this during compilation, but thinking about it more that doesn't really make any sense. I appreciate the help!

closed 2022-07-17 by Scryfox