vhelin/wla-dx #526: assembling gameboy ends after Pass 1...

Strugglemeat, 2022-08-02:
./wla-gb -v -o dragontail.o dragontail.asm
`Pass 1...`
after this it just exits
any advice?

vhelin, 2022-08-02:
No advice without getting my hands on dragontail.asm as something inside that file will break the parsing (pass_1)...

vhelin, 2022-08-02:
You could add .PRINT calls to dragontail.asm here and there and try to pinpoint where exactly the parser quits. If you don't want to share dragontail.asm with us...

Strugglemeat, 2022-08-02:
[dragontail.asm.zip](https://github.com/vhelin/wla-dx/files/9242885/dragontail.asm.zip)

the disassembly was saved from the disassembler of Emulicious

vhelin, 2022-08-02:
Looking at this code

```
SECTION "wram_c000", WRAM0[$C000]
_RAM_C000_: db
_RAM_C001_: db
_RAM_C002_: db
_RAM_C003_: db
_RAM_C004_: db
_RAM_C005_: db
_RAM_C006_: db
_RAM_C007_: db
_RAM_C008_: db

SECTION "wram_c0a0", WRAM0[$C0A0]
_RAM_C0A0_: db
_RAM_C0A1_: db
_RAM_C0A2_: db
_RAM_C0A3_: db

...
```

... I can tell you it's not in WLA's syntax... WLA-GB should give an error parsing the .SECTIONs, so I'll see why it doesn't do that and fix that, but I cannot make this .asm file to be assembled by WLA-GB.

vhelin, 2022-08-02:
Reading the first few lines of dragontail.asm makes me think you want to use RGBASM instead of WLA...

closed 2022-08-02 by vhelin