vhelin/wla-dx #527: Z80 asm jr/jp/ret/call condition codes enhancement

sverx, 2022-08-02:
Well maybe it's just me but I often find myself desiring we had more condition codes than the basic `Z/NZ`, `C/NC` etc. for `JR`/`JP`/`CALL`/`RET` Z80 asm instructions, mainly to make source code clearer.

For instance, take this very simple piece of code:

```
ld a,(some_var)
cp SOME_CONST
jr z,_some_label
jr nc,_some_other_label
```
sure we can tell what's going on here, but I still suspect if we had _enhanced_ condition codes (actually just _aliases_ of what we have already) the code would be clearer.

So I'm probably just asking for feedback on the idea for the moment. My take would be to add three letters condition codes such as
`equ` (equal - alias for `z`)
`neq` (not equal - alias for `nz`)
`gte` (greater than or equal - alias for `nc`)
`lss`/`lst` (or some better 3 letters code for less than - alias for `c`)

... and maybe even more than these might be needed, but those are the ones I would gladly use. :smiley:

vhelin, 2022-08-02:
Well, such aliases are quick and easy to add, and I support anything that'll make asm coding easier :)

sverx, 2022-08-02:
Thanks! Just pick some nice aliases and let's see what others suggest too - as for me, I'm not that fond of `gte` but `lss` / `lst` are the worst ones I think.

sverx, 2022-08-02:
`EQU` `NEQ` `GEQ` `LSS` would match DOS [batch scripting relational operators](https://www.tutorialspoint.com/batch_script/batch_script_relational_operators.htm)

not that I'm that fond of that `GEQ` either but we need to pick something in the end...

vhelin, 2022-08-02:
As an old 68k programmer how these would sound like:

```
jr z, some_label 
->
jeq some_label

jr nz, some_label 
->
jne some_label

jr nc, some_label
->
jgt some_label

jr c, some_label
->
jlt some_label
```

sverx, 2022-08-02:
I would still prefer to keep the instruction separated from the condition, not to mention that also `RET` and `CALL` can be conditional.

Also since the original conditions are either 1 or 2 chars long I would define the new aliases using 3 chars exclusively, and have a couple different spellings for _greater-than-or-equal_ and for _less-than_.

I also thought that even `pos` and `neg` could be added as aliases of conditions `p` and `m` which where never really clear, especially the latter.

So my suggestion is still to create a set of condition aliases so that these would be all valid instructions:

```
ret neq

jr equ, some_label

call gte, some_label

call geq, some_label

jp lss, some_label

jp lst, some_label

jp pos, some_label

jp neg, some_label
```

Kroc, 2022-08-03:
NB: These would also apply to GameBoy/SM83, which doesn't have P/M flags

vhelin, 2022-08-03:
I'm ready to implement these, but does anyone else has something to say about this?

vhelin, 2022-08-03:
@Kroc You gave a tumbs down. Why was that? Any way these could be done better?

sverx, 2022-08-03:
> NB: These would also apply to GameBoy/SM83, which doesn't have P/M flags

I think only the 'valid' aliases would work there?
So `pos` and `neg` won't be valid on the SM83/GB as they would alias to not existing jump/call/ret conditions...

Kroc, 2022-08-03:
> @Kroc You gave a tumbs down. Why was that? Any way these could be done better?

Given that Z80 already has a 2 parameter structure for conditional jumps, this syntax confuses the nature of CPU flags and register values; I can't put my finger on it exactly but it feels semantically confusing. Other architectures have "jple"/"jpge" instructions so I would prefer to use that style of syntax.

sverx, 2022-08-04:
@Kroc Even if I understand your concerns, I do believe that the syntax of `JR`,`JP` and `CALL` is 

`instruction [condition] label`

and for `RET` it's

`instruction [condition]`

and what we would create here would simply be _handy nicknames_ for the **existing** conditions, not new syntax.

vhelin, 2022-08-15:
I hope I got all correct. Next I'll add these to WLA-GB as well...

sverx, 2022-08-16:
> I hope I got all correct. [...]

So far so good.

```
    ; check if bullet.damage >= enemy.HP
    ld a,(iy+enemy.HP)
    sub (ix+bullet.damage)
    jr lss,_kill_enemy            ; if HP < damage
    jr equ,_kill_enemy            ; if HP = damage
    ld (iy+enemy.HP),a
```

vhelin, 2022-08-21:
I guess this issue has been taken care of?

sverx, 2022-08-22:
I already used a few of them a couple of times now and they seem working just fine - though I didn't do any extensive tests.

But of course you can mark this issue as closed. I will let you know later if something breaks :smiley:

vhelin, 2022-08-23:
I think I'll go through all the changes row by row and check that they are ok and if they are then I'll close this issue. But can you confirm that these flag changes are correct?

C -> LSS and LST, M -> NEG, NC -> GTE and GEQ, NZ -> NEQ, P -> POS, Z -> EQU

sverx, 2022-08-23:
As said I've only used a few of them so far, but they seem to work just fine.

The assignments are correct for what I understand.

When doing `CP x` (likewise with `SUB x`) you'll get the flags according to the A - x operation, so:

- Carry Flag Set ( C ) if A < x :arrow_right:  `LSS` / `LST`
- Carry Flag Reset ( NC ) if A ≥ x :arrow_right: `GTE` / `GEQ`
- Zero Flag Set ( Z ) if A == x :arrow_right: `EQU`
- Zero Flag Reset ( NZ ) if A != x :arrow_right: `NEQ`
- Sign Flag Set ( M ) if msb of result is 1 (signed negative) :arrow_right: `NEG`
- Sign Flag Reset ( P ) if msb of result is 0 (signed positive) :arrow_right: `POS`

vhelin, 2022-08-23:
Double checked Z80 and GB-Z80, all seems to be good.

closed 2022-08-23 by vhelin

sverx, 2025-03-05:
I just got my first use case of the missing overflow flag aliases: it's in 16-bit signed comparison.

Let me show an example - this performs a signed comparison of two 16 bits values by performing the operation `BC-(HL)` and then using the overflow and the sign flags to determine if BC<(HL) ,  as described for instance in this [stackexchange page](https://retrocomputing.stackexchange.com/questions/9163/comparing-signed-numbers-on-z80-8080-in-assembly)

```
  ld hl,current_scroll_value
  ld a,c
  sub a,(hl)
  inc hl
  ld a,b
  sbc a,(hl)      ; this performs BC-(HL)
  jp PO,+         ; if not OVR flag, skip XOR
  xor $80
+:
  ret pos         ; return if current_scroll_value <= X coordinate
```
note that it says `jp PO` when in fact there's no parity check involved, as PO means 'not overflow' in this case.

But... yeah, I have no good idea regarding some three letters aliases that would work. Maybe just `V` for overflow and `NV` for no-overflow? I'm not fond of the idea of adding aliases that aren't three letters. 🤔 

What do you think about this?

vhelin, 2025-03-12:
Well, if we want three letter aliases, can't we just make up something like NOF (no overflow) and OFW (overflow)?

I tried to Google if some CPU asm had three letter aliases for "no overflow" and "overflow", but couldn't find any. But my Google skills suck...

sverx, 2025-03-13:
I don't know, and I also dropped that assembly code above for a simpler solution so... no rush.
I'll let you know if by chance I find two good candidates.
Thanks!