vhelin/wla-dx #528: "Error in computation syntax" error when using < or > operators before a - sign (bug?)

sverx, 2022-08-18:
I'm not sure why this happens (or if this is a known problem already) but if I define a value like this:
`.define BOSS_WALK_SPEED $0156`

then when I try to use that define with a minus sign in front, with a > or < operator like this:
`ld (iy+object.int_h_speed),>-BOSS_WALK_SPEED`

I get this error:
```
boss.asm:374: STACK_CALCULATE: Error in computation syntax.
boss.asm:374: ERROR: Couldn't parse "ld".
```

Then, if I change the define to:
`.define BOSS_WALK_SPEED -$0156`
(note the minus sign in front of the value)

and then I do
`ld (iy+object.int_h_speed),>BOSS_WALK_SPEED`

then it just works, but I wonder why I can't use the first syntax, as I doubt that `>-VALUE` could be misinterpreted...

sverx, 2022-08-18:
I forgot - a workaround I can also use is to add parenthesis like this:
`ld (iy+object.int_h_speed),>(-BOSS_WALK_SPEED)`
 
and it works, but still I wonder why I need those.

vhelin, 2022-08-18:
The parser might break when there are two calculation symbols in a row. I'll check it out if I can make an exception for cases like <-A and >-A, have to check if A+-B works as well...

vhelin, 2022-08-19:
I hope the latest sources fix your issue. In fact I found another issue when investigating this: If a pending calculation referenced another pending calculation with a negative sign then the sign was ignored... This might want me to release v10.3 pretty soon...

sverx, 2022-08-20:
Thanks, I'll test this ASAP and will let you know.

closed 2022-08-28 by vhelin