vhelin/wla-dx #529: IF works in a weird way? (bug?)

sverx, 2022-08-18:
The documentation for `.IF` says:
> Operands must be immediate values or strings.

But labels work too. Unfortunately they don't work together with `.DEFINE`s, so if I write

```
.if (some_label_end-some_label) != 64
   .fail "LUT should have 64 entries!"
.endif
```
it  works, but if I do

```
.define LUT_SIZE 64 
.if (some_label_end-some_label) != LUT_SIZE
   .fail "LUT should have 64 entries!"
.endif
```
I will get the following error:

```
LUT_asm:14: INPUT_ERROR: .IF needs immediate data.
LUT_asm:14: ERROR: Couldn't parse ".if".
```

Then I noticed in the documentation example there's this:
`.IF DEBUG == 2 && defined(HELLO) && exists("main.s")`

which means that it should be able to test DEBUG value anyway... so I fail to understand the issue I'm facing.

vhelin, 2022-08-19:
Thanks for the report, I'll try to solve this issue this weekend!

vhelin, 2022-08-20:
This issue was actually due to WLA's delta calculator that wasn't turned on for

(some_label_end-some_label) != LUT_SIZE

but was turned on for

(some_label_end-some_label) != 64

The code that decided when to try to calculate deltas was a bit... simple.

sverx, 2022-08-24:
Fixed! :tada:

closed 2022-08-24 by vhelin