vhelin/wla-dx #531: .INCBIN "file.bin" READ enhancement

sverx, 2022-09-08:
I was wondering if it's possible for `.INCBIN` to load only a certain number of bytes from a binary file, but this number is not known beforehand but it's instead based on the size of the binary file itself.
For instance I want to be able to read everything up to the last byte but not load that last one too, say because it is a termination marker I need to skip to juxtapose some other data.

If it's not already possible (I checked the docs and it doesn't seem to be) I think something like using a negative number for the READ parameter as in

`.INCBIN "file.bin" READ -1`

would do.

Of course using 

`.INCBIN "file.bin" SKIP 4 READ -1`

should still skip 4 bytes at the beginning of the file and 1 at the end, so it's not simply `READ` _size-1_

vhelin, 2022-09-08:
Sorry, cannot check the docs right now, will get back to this issue later today evening, but hmm

isn't there .FSIZE?

Use that to get the file's size and then feed the size-1 to .INCBIN?

If not possible, well, I'll get back to this later today... Should be easy to implement such negative sizes...

sverx, 2022-09-08:
No rush!

BTW it's not just about taking the size and subtracting one anyway...

Thanks!

vhelin, 2022-09-08:
I hope this commit fixes your issue

sverx, 2022-09-08:
I'm yet not using this feature, but I think I soon will.
I will report if something will not be working as expected - in the meantime, thank you! :heart:

vhelin, 2022-09-19:
I think I'll close this issue now as at least my tests pass...

closed 2022-09-19 by vhelin

sverx, 2022-09-20:
> I think I'll close this issue now as at least my tests pass...

Yes, I'll report back in case it won't work as expected. Thanks!