vhelin/wla-dx #534: Allow definition substitutions in file paths

maxim-zhao, 2022-09-30:
I'd like to make this work:

```asm
.include "words.{LANGUAGE}.asm"
```

with `-D LANGUAGE=xxx` on the commandline (i.e. definitely resolvable).

vhelin, 2022-09-30:
Sounds like a solid enhancement! But we are too close to v10.3 release, I want to do that first and then add this as a 1st new feature to v10.4a...

maxim-zhao, 2022-10-02:
As you might otherwise forget, please make this also work for makefile generation...

Kroc, 2022-10-02:
This would need some form of escaping since "{" & "}" are valid filename characters, but isn't filename substitution already possible via string concatenation? i.e. `.include "words.", LANGUAGE, ".asm"`

vhelin, 2022-10-02:
> This would need some form of escaping since "{" & "}" are valid filename characters, but isn't filename substitution already possible via string concatenation? i.e. `.include "words.", LANGUAGE, ".asm"`

That is true, your example works already.

vhelin, 2022-10-02:
Hmm, this seems to work already

        .include { "{INCLUDE}3.s" }

so in the case of this issue

        .include { "words.{LANGUAGE}.asm" }

should work already?

vhelin, 2022-10-02:
Just tested, it should also work with makefile generation unless I am mistaken. @maxim-zhao please report back if you get it to work and thus this issue can be closed...

maxim-zhao, 2022-10-02:
Ah - I mis-read the syntax, the extra curly braces are important! Yes, it works.

closed 2022-10-02 by maxim-zhao