vhelin/wla-dx #535: linker error [bug?] when using LD (iy+d) with a compound displacement

sverx, 2022-10-05:
I just got a pretty weird **linker** error:
`[...]/build/main.o: objects_draw.inc:29: PARSE_STACK: Unresolved reference to "object.VRAM_pool_blocks".`

when I do
`ld e,(iy+object.VRAM_pool_blocks+1) `

this is the object definition:

```
.struct object
    status                       db
[... etc ...]
    VRAM_pool_blocks             dsb 8
.endst
```

Of course when I remove the +1 part it links correctly, so this works fine:
`ld e,(iy+object.VRAM_pool_blocks) `

Weirdly enough in the same project I can do
`ld (iy+object.VRAM_pool_blocks+1), 4`
and the linker doesn't complain. :thinking:

vhelin, 2022-10-05:
I'll try to check this later today in the evening. Can you do this:

ld e,(iy+(object.VRAM_pool_blocks+1))

?

sverx, 2022-10-05:
Same error, I had tried it already.

Also same error with
`ld e,(iy+1+object.VRAM_pool_blocks) `

No rush, I already have a workaround in place.

vhelin, 2022-10-05:
Hmm I'm not able to reproduce this issue. Could you make a minimal source code block that I could use for testing?

I found another issue here, which seems to have been a long time in WLA:

ld e,(iy+n) 

here n is PC relative is n is just an address label

ld e,(iy+n+1) 

here n is not PC relative if n is just an address label -> WLA doesn't configure n+1 to contain PC relative references...

Fix coming to this in a few minutes...

vhelin, 2022-10-05:
I made tests like these, but could not reproduce your issue, FYI:

```
        .db "10>"               ; @BT TEST-10 10 START
address_test_10:
        ld e,(iy+address_test_10) ; @BT FD 5E FD
        ld e,(iy+address_test_10+1) ; @BT FD 5E FB
        ld e,(iy+0+address_test_10) ; @BT FD 5E F7
        ld e,(iy+0+address_test_10+1) ; @BT FD 5E F5
        .db "<10"                 ; @BT END

        .struct Object
        status  db
        data    dsb 8
        .endst

        .dstruct Object1 instanceof Object DATA 1, "HELLO"

        .db "11>"               ; @BT TEST-11 11 START
        ld e,(iy+Object1.data) ; @BT FD 5E F2
        ld e,(iy+Object1.data+1) ; @BT FD 5E F0
        ld e,(iy+Object1.data) ; @BT FD 5E EC
        ld e,(iy+Object1.data+1) ; @BT FD 5E EA
        ld e,(iy-0+Object1.data) ; @BT FD 5E E6
        ld e,(iy-0+Object1.data+1) ; @BT FD 5E E4
        ld e,(iy-0+Object1.data) ; @BT FD 5E E0
        ld e,(iy-0+Object1.data+1) ; @BT FD 5E DE
        .db "<11"                   ; @BT END

        .db "12>"               ; @BT TEST-12 12 START
        ld e,(iy+Object.data) ; @BT FD 5E 01
        ld e,(iy+Object.data+1) ; @BT FD 5E 02
        .db "<12"               ; @BT END

        .section "Section2" force
        .db "13>"               ; @BT TEST-13 13 START
        ld e,(iy+Object2.data) ; @BT FD 5E 07
        ld e,(iy+Object2.data) ; @BT FD 5E 04
        .db "<13"                   ; @BT END
        .ends
```

sverx, 2022-10-05:
> Hmm I'm not able to reproduce this issue. Could you make a minimal source code block that I could use for testing?

I still got the issue with you latest commit. I'll see if I can create test case. Give me a sec.

sverx, 2022-10-05:
Oh wait, I found the culprit and now it's working.

Still it's pretty interesting.

So, the `.struct object` was not yet defined when I was using it in my `ld e,(iy+object.VRAM_pool_blocks+1)` ... but it didn't complain at all anyway and was compiling and working fine until I needed to add that +1 part.

So in your test, try placing the struct definition *after* the code using it. It won't complain as long as you don't do some math on it. :thinking:

vhelin, 2022-10-05:
If you don't define the .STRUCT first WLA will postpone the calculation to WLALINK as .STRUCT kinda just creates .DEFINEs AFAIK. Perhaps pass_4.c can find the .STRUCT unless it's inside a calculation?

This is kinda how WLA works by design, but gotta investigate why it works without the calculation as AFAIK it should fail in both cases...

sverx, 2022-10-05:
Yes, I would have expected it to fail immediately when referencing it before defining it.

That's why I didn't think it wasn't yet being defined in the first place :face_with_spiral_eyes: 

Thanks for the help in any case! :heart:

vhelin, 2022-10-05:
It cannot actually fail until WLALINK processes it as the reference could be in another file.

But yeah, it seems that pass_4.c can use .DEFINEs (and thus .STRUCTs) created later in the file when the reference is not a calculation, but calculations are passed to WLALINK which will fail as the .DEFINEs / .STRUCTs are not EXPORTed...

Perhaps it might make sense to let WLA use .DEFINEs / .STRUCTs that are later defined inside calculations in pass_4.c so WLA could solve the calculation and not pass it to WLALINK?

vhelin, 2022-10-05:
Also thank you for reporting this ~issue as it enabled me to find another issue in WLA. :)

sverx, 2022-10-05:
> Perhaps it might make sense to let WLA use .DEFINEs / .STRUCTs that are later defined inside calculations in pass_4.c so WLA could solve the calculation and not pass it to WLALINK?

makes A LOT of sense in my honest opinion!

vhelin, 2022-10-05:
> > Perhaps it might make sense to let WLA use .DEFINEs / .STRUCTs that are later defined inside calculations in pass_4.c so WLA could solve the calculation and not pass it to WLALINK?
> 
> makes A LOT of sense in my honest opinion!

I also think so. Let me sit on this a few hours, in case I can think of a reason why not to do this.

Also as non-calculations can already be solved in pass_4.c this way, why not calculations as well? Hmm...

sverx, 2022-10-05:
Sure, no rush! And this issue can be closed as it's solved already.

vhelin, 2022-10-06:
Ok, your original test case that caused this issue to be reported

ld e,(iy+object.VRAM_pool_blocks+1)

should now work with the latest sources.

closed 2022-10-06 by vhelin