vhelin/wla-dx #536: .ENUM generates ROM data when referenced .STRUCT has a given SIZE

Ramsis-SNES, 2022-10-10:
I was scratching my head recently when I tried to replace a `.STRUCT` like this:

```
.STRUCT teststruct
	testbyte1	db 1
	testbyte2	db 1
	padding 	dsb 254
.ENDST
```
with this:
```
.STRUCT teststruct SIZE 256
	testbyte1	db 1
	testbyte2	db 1
.ENDST
```
... and wlalink threw up an error message like, `INSERT_SECTIONS: No room for section ... (XXX bytes)`.

Note that `teststruct` was/is only ever referenced by an `.ENUM`, which is supposed to only create definitions and not generate ROM data (unlike e.g. a `.DSTRUCT`).

I dug a bit deeper and tried to come up with a test project in order to reproduce the issue, as my game project has become a bit convoluted by now (and its GitHub repo isn't up-to-date either).

So, to reproduce, open https://github.com/vhelin/wla-dx/blob/master/tests/65816/base_test_3/main.s and add the following in between lines 33 and 35:

```
.STRUCT teststruct SIZE $10000
	testbyte1	db ;dsb $8000
	testbyte2	db ;dsb $8000
.ENDST

.ENUM 0
	test INSTANCEOF teststruct
.ENDE
```
The error message will be:
```
MEM_INSERT: Origin ($10000) overflows from bank (0).
   ^ main.s:42: Writing .DSB data
```
Again, we don't even _want_ to generate ROM data, just definitions. Let's be silly and replace both `db` with `dsb $8000`, and it assembles just fine, as if the `.ENUM` (or the `SIZE $10000`, for that matter) wasn't even there.

In short, it seems that `SIZE` messes up the purpose of `.ENUM` and somehow makes it generate ROM data instead of definitions, but only if the value of `SIZE` is actually higher than the number of bytes in the `.STRUCT` itself (which is the whole point, really).

Thanks for reading, and do let me know if there are any questions. :-)

vhelin, 2022-10-10:
Oof, sounds like I might have misunderstood something (or forgotten something) when implementing that SIZE... I'll try to fix this tomorrow evening.

vhelin, 2022-10-11:
I hope the latest sources solve this issue, and nothing was broken. The .ENUM code is a bit complex IMHO...

Ramsis-SNES, 2022-10-11:
Oh, wow! I'll try right away, and report back!

Ramsis-SNES, 2022-10-11:
It works flawlessly now.
You are a code wizard, Ville! :D
Thanks a lot!

vhelin, 2022-10-11:
> You are a programming wizard, Ville! :D

Thanks, but I just spend too much time coding -> practice, practice, practice... :)

And I make a lot of bugs while coding! Fortunately my manual tests help be quite a lot...

Ramsis-SNES, 2022-10-11:
I spoke too soon (sorry), there's still an issue remaining with nested `.STRUCT`s.

If you add

```
.STRUCT testnest
	nest1 INSTANCEOF teststruct
.ENDST
```
around line 39 in the new https://github.com/vhelin/wla-dx/blob/master/tests/65816/base_test_3/main.s file and make, you'll see in the verbose info that it generates 6 more bytes of ROM.

vhelin, 2022-10-11:
How about now? :) I said I make bugs... :D

Ramsis-SNES, 2022-10-11:
Yay, now we're good! Thank you so much, Ville! :D

vhelin, 2022-10-11:
No, thank you for reporting this bug. If you hadn't reported it now, it wouldn't have gotten fixed today...

Ramsis-SNES, 2022-10-11:
Haha, true. And no false alarm this time, for a change! XD

BTW, you might want to clean up the /tests/65816/base_test_3/main.s file again (I merely chose this one for reproducing the issue because it was small, and a HiROM example, like my game project).

vhelin, 2022-10-12:
Done, I moved the .STRUCT tests to struct_tests...