vhelin/wla-dx #538: Issue with nested unnamed .STRUCTs

Ramsis-SNES, 2022-10-12:
If you nest unnamed `.STRUCT`s like this:

```
.STRUCT 1 SIZE 8
    testbyte1   db
.ENDST

.STRUCT 2 SIZE 8
    testbyte2   db
.ENDST

.STRUCT both
    INSTANCEOF 1    ; name1 INSTANCEOF 1
    INSTANCEOF 2    ; name2 INSTANCEOF 2
.ENDST

.ENUM 0
    test    INSTANCEOF both
.ENDE

```
... assembly fails with `DIRECTIVE_ERROR: "_paddingof_both" was defined for the second time.`

However, if you do put names in front of the `INSTANCEOF`s (in `.STRUCT` "both"), it works fine, though there are instances (pun intended) where you'd want to omit the names.

vhelin, 2022-10-12:
Oof, it seems that _paddingof_* generator doesn't handle unnamed .STRUCTs... I'll try to fix this today evening (i.e. in 3-6 hours)...

Ramsis-SNES, 2022-10-12:
Yep, seems to work alright! Thanks! :D

closed 2022-10-12 by Ramsis-SNES