vhelin/wla-dx #540: Add max(a, b) and min(a, b) that can be delayed to WLALINK

vhelin, 2022-10-24:
The title says it all.

maxim-zhao, 2022-10-25:
I'll explain my use case as it may help trigger a better solution. It gets complicated...

I am using WLA DX features to define a range of definitions for a sequence of arrays of bytes in memory. The size of each of these arrays is dependent on the values of some data in a YAML file, which I process into `.include`d asm using a Python script. The exact values depend on a compile time setting (specifically, it's a language selection for a translation). As part of the arithmetic I want to align one such array to align immediately after two others, which may overlap; so I want to align it to the max of the end of the two preceding arrays.

Thus I have an `.if` statement to implement the logic. However, in makefile generation mode, the `.include`d asm file does not exist - the makefile generation will act to trigger it to get generated as needed, based on other rules in my makefile. Therefore the `.if` cannot resolve the condition, so the build fails.

It's somewhat analogous to if you were to use makefile generation in GCC a bit like this:
foo.c:
```c
#include "a.h"
#ifndef bar
#include "b.h"
#endif
```
If a.h contains `#define bar` then b.h is a dependency of foo.c. But what if a.h doesn't exist at makefile generation time?

maxim-zhao, 2022-10-25:
I came up with a creative workaround...
```
.function max(a,b) floor(a/b)*a+floor(b/a)*b
```
This seems to work!

vhelin, 2022-10-25:
Hmm is my math broken? :)

max(10, 5) = floor(10/5)*10 + floor(5/10)*5 = 2*10 + 0*5 = 20 ... ?

maxim-zhao, 2022-10-25:
Only when a is close to b...

maxim-zhao, 2022-10-25:
A later thought: here I am dealing with addresses in system RAM on Master System, so in the range $c000..$dfff. Thus they are guaranteed to be close enough to each other that my formula above is OK. For many, many other cases it's a terrible idea :)

vhelin, 2022-10-25:
Hopefully soon you should get working built-in functions min(a,b) and max(a,b)... :)

maxim-zhao, 2022-10-25:
At which point I will find out what happens when I have a name collision with a `.function`...

vhelin, 2022-10-25:
That is actually a good point; I don't think WLA checks if you try to create a .define with the same name as built-in functions, but built-in functions should override user defined ones when they are used. I'll add checks to .function for name clashes...

vhelin, 2022-10-25:
Ok, fixed!

vhelin, 2022-10-25:
Do min(a,b) and max(a,b) work for you?

maxim-zhao, 2022-10-27:
Yes, as does the error message when you have a conflicting name. Thanks!

closed 2022-10-27 by vhelin