vhelin/wla-dx #541: Incorrect high and low byte of address

lajohnston, 2022-11-05:
Hi there,

I have a strange issue with the latest `master` version, which has been working fine on previous versions up until now. I have a label in a section and if I load it into a 16-bit register, it works fine, but if I try to load just the high byte or low low byte it produces a different and incorrect result. i.e.
```
ld iy, myLabel      ; correct value ($08FF)
ld iyh, >myLabel    ; incorrect high byte ($D6)
ld iyl, <myLabel    ; incorrect low byte ($D0)
```
Has anything changed in this area that would affect this? It also produces the same incorrect result if I use `hibyte` and `lobyte`. I'm not sure if I can create a simplified example but I can provide some binaries and source code to replicate if needed.

vhelin, 2022-11-06:
Thanks for letting me know! I did touch the code where high and low bytes are calculated, but in the worst case if the new code failed it should give negative values. I'll look into this issue right thing tomorrow. And I'll clearly need to improve my tests as they are passing currently...

Ramsis-SNES, 2022-11-06:
I just tried assembling my project, and while everything worked fine a few weeks ago, I'm now running into issues possibly related to this one, so I'll just comment here (sorry if this turns out to be another issue).

WLA DX is suddenly giving me weird error messages relating to `lobyte()`, `loword()`, and `bankbyte()`, e.g.:

```
lib_macros.asm:190: INPUT_NUMBER: EXPAND_MACRO_ARGUMENTS: Macro argument \2 is a pending stack calculation and cannot be expanded into a string.
lib_macros.asm:190: STACK_CALCULATE: Unhandled result type 0 of a in bankbyte(a)!
```

The line of code in question reads `lda #bankbyte(\2)`, and `\2` simply references a ROM label (supplied as a macro argument, obviously).

Another error message says:

```
lib_macros.asm:186: PARSE_STACK: Unresolved reference to "8576".
```
Here, the line of code reads `lda #lobyte(\3)`, where `\3` references a 16-bit SNES B-bus register with a `.DEFINE`d label (possibly $2180 in this case = hex value of 8576).

Thanks! :-)

vhelin, 2022-11-06:
> Has anything changed in this area that would affect this? It also produces the same incorrect result if I use `hibyte` and `lobyte`. I'm not sure if I can create a simplified example but I can provide some binaries and source code to replicate if needed.

That source code would be really useful as I cannot repeat this bug on my computer. Maybe the bug is elsewhere and needs more code to happen?

vhelin, 2022-11-06:
If you guys have extra time, could you pinpoint the commit which after you started to see these errors.

vhelin, 2022-11-06:
@Ramsis-SNES I did this and it works just fine:

[image]

Could you also show some code that doesn't work, I don't think I'm properly replicating your issue...

lajohnston, 2022-11-06:
Ok it looks like the following replicates my issue. It seems the problem is restricted to the Z80 8-bit iyh, ihl, ixh, ixl registers as it works fine using D and E.

```
.memorymap
defaultslot 0
slotsize $8000
slot 0 $0000
.endme

.rombankmap
bankstotal 1
banksize $8000
banks 1
.endro

.bank 0 slot 0
.org $0000
.section "boot" force
    di              ; disable interrupts
    im 1            ; Interrupt mode 1
    ld sp, $dff0

    ld iy, myLabel      ; correct
    ld ixh, >myLabel    ; incorrect
    ld ixl, <myLabel    ; incorrect

    ld hl, myLabel      ; correct
    ld d, >myLabel      ; correct
    ld e, <myLabel      ; correct

    -: jp -
.ends

.org $0066
.section "pause" force
    retn
.ends

.section "someSection" free
    myLabel:
        outi
        ret
.ends
```

I'll try with other commits to see if I can pin point when it starts

lajohnston, 2022-11-06:
Ok, it looks like the issue starts with 240e2bc9

vhelin, 2022-11-06:
Great detective work! I'm starting to think that the change in that commit is really something that needs to be reversed! I'm guessing I misunderstood the whole thing in that commit now that I think of it...

Also @Ramsis-SNES I'm thinking that your issue is not the same that @lajohnston is seeing here... @Ramsis-SNES I just made a commit. If it passes the CI tests, could you try it out if it fixes your issue?

lajohnston, 2022-11-06:
Great, thanks! I've compiled latest `master` since the revert and the issue is now fixed using the above example :)

vhelin, 2022-11-06:
I really need to stop the habit of going through the code and spotting "errors" where none exists. :D

Ramsis-SNES, 2022-11-06:
I just tried it with the latest sources, and while my issue persists, something did indeed change: The `bankbyte()` thing now yields `lib_macros.asm:190: PARSE_STACK: Unresolved reference to "8265728".` (Weird because that value equals $7E2000, which isn't even 8-bit. :D )

I tried the macro tests to replicate the issue but wasn't able to either, so I tried the trusty HiROM example instead, and boom, the issue thankfully pops up. Just temporarily replace `main.s` in https://github.com/vhelin/wla-dx/tree/master/tests/65816/base_test_3 with the following file (zipped as GitHub doesn't support plain text uploads):

[main.s.zip](https://github.com/vhelin/wla-dx/files/9946717/main.s.zip)

Trying to make it should yield `main.s:39: PARSE_STACK: Unresolved reference to "8472".`

Thanks for your efforts! :-)

vhelin, 2022-11-06:
Thanks @Ramsis-SNES I can repeat the bug with that main.s.zip! I also get that "8472". It's getting late here so I don't think I can fix this today, but will continue tomorrow...

vhelin, 2022-11-06:
I was able to locate the problem, and I hope the latest commit fixes the issue. Recent changes how lobyte() and other built-in functions are handled put more pressure on expand_macro_arguments() and it just couldn't handle the new requirements... I hope it does now.

Ramsis-SNES, 2022-11-07:
I tried it, my issue is fixed. Thanks a lot, @vhelin! :D

vhelin, 2022-11-07:
No problem, no need to thank me for fixing an issue I created myself, LOL. :D And thank you guys for reporting these issues! Now we have better automated tests that should prevent these things breaking again in the future...

vhelin, 2022-11-07:
I'll close this issue as it seems that both of the issues were solved. If you spot any more issues, please report them asap so I can fix them! :)

closed 2022-11-07 by vhelin