vhelin/wla-dx #542: Intermittent internal errors

maxim-zhao, 2022-11-10:
I have a CI build on AppVeyor for [a project](https://github.com/maxim-zhao/psrp).  It has started to intermittently produce build errors that are resolved by a re-build using the exact same source and WLA DX binary. Example:

https://ci.appveyor.com/project/maxim-zhao/psrp/builds/45340429 - failed:
```
wla-z80.exe -M -D LANGUAGE=pt-br -o generated/ps1jert.pt-br.sms.o ps1jert.sms.asm > generated/auto.makefile.pt-br
wla-z80.exe -M -D LANGUAGE=es -o generated/ps1jert.es.sms.o ps1jert.sms.asm > generated/auto.makefile.es
wla-z80.exe -M -D LANGUAGE=en -o generated/ps1jert.en.sms.o ps1jert.sms.asm > generated/auto.makefile.en
wla-z80.exe -M -D LANGUAGE=literal -o generated/ps1jert.literal.sms.o ps1jert.sms.asm > generated/auto.makefile.literal
data_stream_parser_parse(): Unknown internal symbol "0". Please submit a bug report!
data_stream_parser_parse(): Unknown internal symbol "0". Please submit a bug report!
data_stream_parser_parse(): Unknown internal symbol "0". Please submit a bug report!
data_stream_parser_parse(): Unknown internal symbol "0". Please submit a bug report!
wla-z80.exe -M -D LANGUAGE=ca -o generated/ps1jert.ca.sms.o ps1jert.sms.asm > generated/auto.makefile.ca
data_stream_parser_parse(): Unknown internal symbol "0". Please submit a bug report!
data_stream_parser_parse(): Unknown internal symbol "0". Please submit a bug report!
```
https://ci.appveyor.com/project/maxim-zhao/psrp/builds/45340463 - no issue

Both of these are using WLA DX version + commit hash prefix:

* wlalink 5.19a (28.10.2022) ed05e90 
* wla-z80 10.4a (31.10.2022) ed05e90 

The build is using `make -j` so multiple invocations are active at once, could that be related? It is also a "clean" build every time, no files left from earlier builds, I even redownload the WLA DX binaries as part of the build as it's a fresh VM.

I then updated to latest WLA DX:

* wlalink 5.19a (9.11.2022) 71160c6 
* wla-z80 10.4a (9.11.2022) 71160c6 

https://ci.appveyor.com/project/maxim-zhao/psrp/builds/45340676
```
wla-z80.exe -M -D LANGUAGE=literal -o generated/ps1jert.literal.sms.o ps1jert.sms.asm > generated/auto.makefile.literal
wla-z80.exe -M -D LANGUAGE=en -o generated/ps1jert.en.sms.o ps1jert.sms.asm > generated/auto.makefile.en
wla-z80.exe -M -D LANGUAGE=es -o generated/ps1jert.es.sms.o ps1jert.sms.asm > generated/auto.makefile.es
wla-z80.exe -M -D LANGUAGE=de -o generated/ps1jert.de.sms.o ps1jert.sms.asm > generated/auto.makefile.de
wla-z80.exe -M -D LANGUAGE=ca -o generated/ps1jert.ca.sms.o ps1jert.sms.asm > generated/auto.makefile.ca
ps1jert.sms.asm:5024: FIND_FILE: Could not open "generated\tree.ca.asm", searched in the following directories:
./ (current directory)
ps1jert.sms.asm:5024: ERROR: Couldn't parse ".include".
wla-z80.exe -M -D LANGUAGE=fr -o generated/ps1jert.fr.sms.o ps1jert.sms.asm > generated/auto.makefile.fr
wla-z80.exe -M -D LANGUAGE=pt-br -o generated/ps1jert.pt-br.sms.o ps1jert.sms.asm > generated/auto.makefile.pt-br
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
ps1jert.sms.asm:4682: ERROR: Section with ID "221" has gone missing! Please submit a bug report!
```
All these have the -M parameter so a missing file ought not to be reported.

Attempt 2: https://ci.appveyor.com/project/maxim-zhao/psrp/builds/45340717 - success
Attempt 3: https://ci.appveyor.com/project/maxim-zhao/psrp/builds/45340742 - success
Attempt 4: https://ci.appveyor.com/project/maxim-zhao/psrp/builds/45340778 - success

I have very rarely been able to produce the same errors when building locally, and re-building always fixes it. I built the whole project from scratch 100 times in a loop and it failed two times.

Others working on the project have reported similar errors across many recent WLA DX versions. The specific code I built in these tests is https://github.com/maxim-zhao/psrp/commit/2c0dd7d96e073a9fa2c6632c27b9b6bf25ea501a

vhelin, 2022-11-10:
As a quick note, cannot investigate this right now as I'm working, the only times I've had locally similar problems (not "Unknown internal symbol"s though) is that when I change igb.c etc. files that contain the instruction structs as after a change there a total rebuild for that CPU is required. Also I got the section missing error locally when not reassembling WLA generated object files after some recent changes in WLA DX, or optimizations to be precise.

I'll get back to this later today after work or latest tomorrow.

maxim-zhao, 2022-11-10:
I think I have diagnosed it and it is indeed caused by my efforts to maximise my build parallelism. 

WLA DX has a function `generate_tmp_name()` which tries to open a file named `wla_tmp_%d.tmp` for increasing integer values, in order to discover an unused filename (when `fopen()` returns NULL). It then passes the filename back to the caller, who generally then opens the file for writing - although maybe doing this later on.

If you have two builds happening at the same time, they will then have a race between finding an unused filename and then writing to it. They might even end up appending each other's file, causing much confusion and build errors.

I'm not sure how best to fix it - standard C doesn't have a way to fail `fopen()` if the file already exists until C2021. Could you use `tmpfile()` instead?

vhelin, 2022-11-10:
Good work! Now we need a way to generate tmp file names that would work in a situation like that.

I think I read somewhere that using tmpfile() is not recommended. ?

Perhaps if we had a true random number at that point we could use it in the tmp file name and there'd almost never be clashes? I mean if fopen() succeeds for such a file name then just take a new random number... ?

maxim-zhao, 2022-11-10:
You need to reach into the realms of GUIDs to have great certainty of uniqueness. Maybe a better approach would be to use the PID in the filename?

vhelin, 2022-11-10:
I think we used PID here previously, but only on some platforms as not all supported platforms provide that GetPID()... Later I streamlined the function to use a running number, and didn't see this issue coming up... What I'd like to have is a solution that would work on every platform, and the same way if possible.

maxim-zhao, 2022-11-10:
How about PID if available, and random number otherwise?

maxim-zhao, 2022-11-10:
I found the change - d36cdfa7d49a508c842c05dfce6169e5965938f8 - and to be honest it seems better before - it applies an upper bound to the integer which is nice. The only argument against it is the complexity of platform-specific `getpid()` equivalents and having things different on every platform.

Another approach which might work would be to use a lock file to ensure serialised access to the function in question.

vhelin, 2022-11-10:
Actually I'm looking at the code and trying to use that tmpfile() instead of avoiding it for some reason I cannot remember - perhaps I was wrong and it could work here?

cr1901, 2022-11-10:
I'm not sure that there is _any_ way in ANSI C alone to atomically coordinate access to the same resource. Using a random number (_with a fixed seed, please_ :P) seems fairly safe in practice though, as `tmpfile` [may or may not](https://tldp.org/HOWTO/Secure-Programs-HOWTO/avoid-race.html) have race conditions from the POV of competing WLA processes.

vhelin, 2022-11-10:
It seems I got tmpfile() to work here locally.

If you use a fixed seed to a pseudo random number generator, won't you get the same series of numbers out of it? I guess we'd need totally random numbers here...

I suggest that I push that tmpfile() change into Git, and if we the reported issue still exists on some system we'll come back to this and find another solution.

cr1901, 2022-11-10:
>If you use a fixed seed to a pseudo random number generator, won't you get the same series of numbers out of it? I guess we'd need totally random numbers here...

Yea I messed up. What I meant was "if there happens to be a bug, there should be a way to duplicate the conditions, so we'll somehow need to know the seed of the RNG used to create temp file names".

vhelin, 2022-11-10:
True that. Let's hope the tmpfile() fix works so well we don't have to spend any more time on this. :) Things can only get more complicated if tmpfile() isn't enough...

maxim-zhao, 2022-11-11:
Issues with `tmpfile()`:
* Potentially still not race-proof, depending on the implementation
* You can't determine what filename it has in order to report it to the user on error (on modern Linux it's never added to the filesystem anyway)
* You can't make it not get deleted when the program exits, e.g. so someone could send it to you with a bug report
  * Although you could add some code to copy it from the temp file into a known file when failing - but that won't help on a crash
* Some versions of Windows + MSVC reportedly create temp files in the system root (c:\) and thus fail when the user does not have write access to that directory. For me on Windows 11 with MSVC v143 it seems not to be an issue (the file is in `%temp%`) but the docs still say that's where the file goes. Maybe the mingw version will do the wrong thing?

If using random numbers you could try to make the commandline args determine the RNG seed, e.g. by hashing them?

Having said that: I ran the same test (1000 parallel builds in a row) with the latest code and this time got no errors. So LGTM, thanks!

closed 2022-11-11 by maxim-zhao

vhelin, 2022-11-11:
I have to say that I've really only once or twice looked inside the tmp file WLA generates when I've encountered an error so not having it in the case of an error is not a big deal for me.

That argument hashing sounds like a good trick if we ever have to create the tmp file name ourselves!

But yeah, let's keep our fingers crossed that tmpfile() is properly implemented on all the platforms WLA gets run on. :)