vhelin/wla-dx #548: Incorrect output for .dsb with calculated values

maxim-zhao, 2022-12-02:
Example code:
```
.dsb 16, <label
.dsb 16, <definition
.dsb 16, definition & 0xff
.dsb 16, 1 + 2

.define definition $6789
label:
```
Assume this is assembled at `.org 0` so `label` = $0040. I would expect to see (hex):
```
40 40 40 40 40 40 40 40 40 40 40 40 40 40 40 40 
89 89 89 89 89 89 89 89 89 89 89 89 89 89 89 89 
89 89 89 89 89 89 89 89 89 89 89 89 89 89 89 89 
03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 
```
I actually get:
```
CD CD CD CD CD CD CD CD CD CD CD CD CD CD CD 40 
89 CD CD CD CD CD CD CD CD CD CD CD CD CD CD CD 
89 CD CD CD CD CD CD CD CD CD CD CD CD CD CD CD 
03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 C3
```
Which looks like three different things are happening...

WLA DX build is git hash efdaca2.

vhelin, 2022-12-02:
Argh, it seems that all .DS*'s pospone only one pending calculation to WLALINK per .DS* - a clear design flaw. Pretty amazing how long this issue has existed - from the very beginning. I'll try to fix this asap!

maxim-zhao, 2022-12-02:
Works for me! 

Related change that needed this: https://github.com/maxim-zhao/psrp/commit/a24767743138ae15386519eea7d0b36a424aa85c#diff-9dfd7fce8825c26216814310a66cb0c7af423e2deace08f3b4e5370b60f132dc

closed 2022-12-02 by maxim-zhao