vhelin/wla-dx #550: "Preferred" banks for .superfree

maxim-zhao, 2022-12-19:
On some platforms, certain ROM banks are more "precious" than others. For example, it's common to have the first 1-2 banks always available (unpaged) and others selected on demand. We use `superfree` to locate sections in these higher banks, but the bank selector seems to prefer to allocate memory bottom-up (subject to prioritisation), so we usually end up fully-packing the lower banks - and then it's unclear how much "free space" we have in the unpaged area.

It'd be helpful therefore to prefer not to put "can go anywhere" sections in the lower banks unless there's no room anywhere else. I guess maybe this could go with the `.slot` directive, e.g. `.slot 2 bank 2..31` to say "any superfree sections go in slot 2, banks between 2 and 31". Expressing more complicated bank choices would I guess require more complicated syntax.

It'd also help with cases where we want certain types of data freely-allocated within a finite number of banks. I know of one commercial game that expresses certain data addresses as 16-bit offsets from a certain point in ROM, and it could be helpful to have "superfree" data sections restricted to the equivalent 4.99-bank range - but I guess that is even more complicated...

vhelin, 2023-01-09:
So we need hmm PREFERREDBANKS and ALLOWEDBANKS to be added to .SECTION definition, and both take a list of banks, e.g. 1-3/4/6/8-10. I think, to make things more clear, we should introduce a new section type that'd accept these keywords and bank lists. Perhaps we should call it "BANKED"?

I might be overengineering here, but should we also make it so that the banks in the list are tried in the given order? So if the list is 1-3/4/6/8-10 then we'll 1st try banks 1,2 and 3, then bank 4, then bank 6 and then banks 8,9 and 10? If the list was 10/2/7 we'd try bank 10 1st, then bank 2 and then bank 7...

maxim-zhao, 2023-01-13:
I'm not sure we need to distinguish between preferred and allowed banks - it's really just the latter, because "preferred" can mean it ends up anywhere so it offers little benefit. Likewise therefore the ordering of banks should also not matter. It seems worthwhile to remove this dimension from the section bank allocator as I'm sure it's already complicated enough.

For my usage, I'd generally want superfree banks to go in banks 2..max, or for the game I mentioned (Sonic the Hedgehog, a little-known hidden gem) n..n+4 where n is fixed. Occasionally I'd want to specify 0..1 for code to go in the "unswapped" zone. So I don't need this complexity - but maybe you can see cases where it's needed?

vhelin, 2023-01-14:
Ok, let it be ALLOWEDBANKS or just BANKS only, no PREFERREDBANKS.

About ordering, I think I should go all the way and have the ordering there just in case someone needs it. If you specify BANKS 2-MAX then banks 2,3,4... are tried in that order. If you specify MAX-2 then banks ...,4,3,2 are tried in that order.

vhelin, 2023-01-14:
Ok, let me know how this new BANKED .SECTION works for you! The parser assumes there are max 1024 banks in the project - is that a problem (should I make it accept any number of banks?)?

vhelin, 2023-01-14:
So if you want the .SECTION to be placed in banks 0-3 or bank 10 or 11 if those are full:

.SECTION "MyBankedSection" BANKED BANKS 0-3/10/11

maxim-zhao, 2023-01-17:
I think `BANKED` is not the best word here - one use case is for "not banked" sections. They're somewhere between `FREE` and `SUPERFREE`... `SEMISUPERFREE`? Naming is hard...

vhelin, 2023-01-17:
Naming surely is hard. :) SEMISUPERFREE sounds good, but would just SEMIFREE be enough?

maxim-zhao, 2023-01-22:
I used it in https://github.com/maxim-zhao/psrp/commit/172bd5917605b14c74293cecab6854ee6b04a54f - in this codebase, bank 0 is precious (the only unpaged area so important functions need to live here), bank 1 is important (nearly always available, paged only in certain circumstances) and bank 2 is also important (the entire script must fit here). But I have loads of superfree things that'd make it hard to see what space is free:
```
ROM bank 0 (66 bytes (0.40%) free)             
ROM bank 1 (144 bytes (0.88%) free)            
ROM bank 2 (13 bytes (0.08%) free)             
...
ROM bank 30 (7962 bytes (48.60%) free)
ROM bank 31 (16384 bytes (100.00%) free)
```
So I simply made all my paged data `semisuperfree banks 3-31` and now I see:
```
ROM bank 0 (226 bytes (1.38%) free) 
ROM bank 1 (872 bytes (5.32%) free) 
ROM bank 2 (1408 bytes (8.59%) free)
...
ROM bank 30 (5797 bytes (35.38%) free)
ROM bank 31 (16384 bytes (100.00%) free)
```
So I'd say this is looking good, thanks!

closed 2023-01-22 by maxim-zhao