vhelin/wla-dx #571: `APPENDTO` adding in reverse order?

Kroc, 2023-04-09:
Starting with

```asm
; each game defines an address to call to start the game;
; this table will be appended to by each game included
;
; the table is aligned to $xx00 so that the game ID
; can easily be mapped to the position in the table
;
.SECTION "InitAddr"                     ALIGN 256 RETURNORG
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
addr_init:
;===============================================================================
        .ADDR           $0000           ;=GAME_NONE
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
.ENDS

; include the games:
; the order of these includes determines their IDs!
;
.INC    "game_acesup.wla"       NAMESPACE "acesup"
.INC    "game_fortunes.wla"     NAMESPACE "fortunes"
.INC    "game_klondike.wla"     NAMESPACE "klondike"
```

And each game appends to the table thusly:

```asm
; add the game's init address to the table
; used for starting games (see "game.wla")
;
.SECTION "InitAddr_<GameName>"          APPENDTO "*:game.InitAddr" RETURNORG
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
addr_init_<gamename>:
        .ADDR           init
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
.ENDS
```

The symbol file indicates that the table is being built in reverse order, after the initial value!

```
00:2c00 game.addr_init
00:2c00 SECTIONSTART_game.InitAddr
00:2c02 klondike.addr_init_klondike
00:2c04 fortunes.addr_init_fortunes
00:2c06 acesup.addr_init_acesup
00:2c08 SECTIONEND_game.InitAddr
```

Assembled using a binary compiled from latest commit, Windows 10, VS2022.

vhelin, 2023-04-09:
This used to work before?

vhelin, 2023-04-09:
According to WLALINK's code sections are sorted by priority first and size (biggest 1st) then so you could fix the issue by giving those sections priorities. There is no "append to sections in the order parsed by the assembler" mode in WLA AFAIK...

vhelin, 2023-04-09:
Also, if you give "-nS" (don't sort the sections) to WLALINK then you might get here what you want...

Kroc, 2023-04-09:
Thank you for the insights! I shall try these out. I don't want to manually order priorities since the very idea is that the files could be included in any order with some enabled or disabled and everything will automatically adjust. The indices would be moved to a variable rather than an `ENUM` of course but I was just testing out building a simple table.

Kroc, 2023-04-09:
Disabling sort isn't viable here due to a large number of aligned RAM sections that overflow RAM if not sorted. I'll have to try priorities. I would argue that appends should always be in order otherwise it makes it impossible to dynamically build tables reliably and it kind of works against the nature of the word "append"; if this were code instead of data it would be impossible to make it work! My Elite disassembly (CA65), for example, has to use appending to sections in-order due to the nature of the very complex internal structure of that program and because it can be built in multiple different ways.

vhelin, 2023-04-10:
As a quick hack, couldn’t you use a .DEFINEd variable for the priority?
Every time a .SECTION would use it you’d .REDEFINE -1 the var…. ?

Kroc, 2023-04-10:
Very, very good idea, thank you!

Kroc, 2023-04-12:
Using a variable for priority worked fine thank you, but upon starting a second table the section is allocated incorrectly, placing it at $0001 where there is already a section placed there by the link script:

```
BANK    0   SLOT    "ROMLO"     ORGA $00    OVERWRITE   "RSTJumpTable"
```

Here's the two sections defined to start jump tables:

```asm
; each game defines an address to call to start the game;
; this table will be appended to by each game included
;
.SECTION "AddrInit"                     RETURNORG
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
addr_init:
;===============================================================================
        .ADDR           $0000           ;=GAME_NONE
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
.ENDS

; each game defines an address to call when the player taps on a card
; this table will be appended to by each game included
;
; each game defines an address to call to start the game;
; this table will be appended to by each game included
;
.SECTION "AddrAction"                   RETURNORG
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
addr_action:
;===============================================================================
        .ADDR           $0000           ;=GAME_NONE
;:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
.ENDS
```

The first section is assigned an address correctly (typically `$C5`). Once the second section is added, despite taking the same form as the first, it is assigned to `$01`!

```
00:00c5 SECTIONSTART_game.AddrInit
00:00c5 game.addr_init
00:00cd SECTIONEND_game.AddrInit
00:0001 SECTIONSTART_game.AddrAction
00:0001 game.addr_action
00:0005 SECTIONEND_game.AddrAction
```

Adding `SEMIFREE` to the second section fixes this, but shouldn't be necessary!  
(especially considering that the first section didn't need it)

```
.SECTION "AddrAction"                   SEMIFREE RETURNORG
```

Side note, adding `SEMIFREE` to the link script instead *doesn't* resolve the issue 😕 

```
BANK    0   SLOT    "ROMLO"     "game.AddrInit"
BANK    0   SLOT    "ROMLO"     SEMIFREE "game.AddrAction"
```

vhelin, 2023-04-12:
This kinda sounds like a separate issue. .SECTION "AddrAction" takes two bytes so it could be placed in any small opening in the ROM available. I don't think I'm seeing the big picture here...

Kroc, 2023-04-12:
Yes, but address 0 is already filled by a section! Once I've appended enough to the section it goes to the right place, but it will get the address wrong if the section size is small enough to fit where RST 0 is already filled

vhelin, 2023-04-13:
Ok, now I get it. I’ll try to replicate this issue today evening and fix it…

Kroc, 2023-04-13:
Sorry, you're right that this is a separate issue! There may be a unrelated off-by-one error lurking somewhere as well because it often select $xxx1 instead of $xxx0 addresses as the first in the bank.

On the subject of append, what are your thoughts on making `APPENDTO` always append sections in order of occurrence? I'm not sure I can see the purpose of an append whose order is largely unknown given that the size of an appended section is not an indicator of required order

vhelin, 2023-04-15:
I tried, but I cannot create an example based on this information that would make .SECTION overlapping to happen. Is there any way you could create a small example that I could take a look at?

> On the subject of append, what are your thoughts on making APPENDTO always append sections in order of occurrence? I'm not sure I can see the purpose of an append whose order is largely unknown given that the size of an appended section is not an indicator of required order

Well, as APPENDTO can be used from multiple files, targetting a single .SECTION somewhere, having such feature as the default might be a bit confusing. But if we would allow that to happen inside just one file it'd be easy to introduce a keyword like AUTOPRIORITY, LEXICALPRIORITY, etc. that'd do the same as this:

> As a quick hack, couldn’t you use a .DEFINEd variable for the priority? Every time a .SECTION would use it you’d .REDEFINE -1 the var…. ?

vhelin, 2023-04-27:
Just for the record, I tried again to create a test case where .SECTIONs would overlap, but didn't succeed.

vhelin, 2023-04-28:
Wait a second, I must be blind. Your link script section is OVERWRITE!

[image]

I think what happens here is that 1st your .SECTION "AddrAction" goes into $0001, and then at last it's overwritten by that "RSTJumpTable" OVERWRITE .SECTION. Why it's OVERWRITE and not FORCE?

Kroc, 2023-04-28:
Sorry being late getting back to you here, busy with child care. The overwrite is there because a safety stub is already written to the RST addresses! https://github.com/Kroc/impatience/blob/master/src/gb/gb_irqs.wla The sections then overwrite to allow them to be rearranged to other RST points if desired. Sorry, this is probably my fault, there's a lot going on in my codebase

vhelin, 2023-04-28:
> -- edit: the linked file shows the RST stubs are written without a section, so I was hoping that they would be marked as 'filled' already when the sections are then arranged during linking

That's how it should work, and I also tested with such code and couldn't get a .SECTION with APPENDTOs to go over such code, they all the time avoided data/code without a .SECTION. Now that I know more about your case, maybe I'll just try again and try to simulate your situation better.

vhelin, 2023-04-28:
Just thinking aloud here, in your example the RST code block has holes in it as RET takes one byte and then the next .ORG is 8 bytes away from the previous:

  - Free space at $0001-$0007 (7 bytes)
  - Free space at $0009-$000f (7 bytes)
  - Free space at $0011-$0017 (7 bytes)
  - Free space at $0019-$001f (7 bytes)
  - Free space at $0021-$0027 (7 bytes)
  - Free space at $0029-$002f (7 bytes)
  - Free space at $0031-$0037 (7 bytes)
  - Free space at $0039-$1fff (8135 bytes)

If .SECTION "AddrAction" is 7 bytes or less it can go into one of those holes, and then later your OVERWRITE .SECTION "RSTJumpTable" will overwrite the whole area and "AddrAction" as well... Could this be the case?

Kroc, 2023-04-28:
That sounds like it! I wasn't aware that the OVERWRITE section would be last -- I assumed it was first; the section names are largely very confusing and it's hard to manage a mental model of them, in which case it's my fault.

vhelin, 2023-04-28:
Yeah, the .SECTION type names have kinda gotten out of hands during the years as we've added more of them... 

I could add aliases for the .SECTION type names, just need to come up with better ones. All suggestions are welcome. :)

Please report back if those NOPs fixed your problem!

vhelin, 2023-04-28:
Also, based on this report, maybe we should add a way to change the order in which the different .SECTION types are written into output? Just thinking, would it even better to do this independently per .SECTION?

Kroc, 2023-04-28:
Ultimately, it's not the fault of `APPENDTO`; perhaps "append" adds confusion because it implies a fixed order, but really what I'm doing is using append as a kludge for the fact I need to build a data table piecemeal. If there were a way to append new data to the end of a table in the order of occurrence.

Better section names would be good, I'll give it some thought. I've rarely used section types simply because they're so hard to understand. I write in a style where the majority of code is not in a section and then use sections only for parts that have to be aligned; this is why `RETURNORG` is such a nuisance to me because it's redundant when I'm trying to avoid gaps in the assembly in the first place

vhelin, 2023-04-28:
>Ultimately, it's not the fault of APPENDTO; perhaps "append" adds confusion because it implies a fixed order, but really what I'm doing is using append as a kludge for the fact I need to build a data table piecemeal. If there were a way to append new data to the end of a table in the order of occurrence, this would solve my use case.

I recently added AUTOPRIORITY to .SECTIONs, which practically does what you want. This works the same way as the .DEFINE (-1) hack for PRIORITY without breaking any existing project. I know this is extra work for the programmer, but there might exist projects that would break if we changed the default behavior of APPENDTO.

We also have .ALIGN in the latest sources, have you checked it out? It works without a .SECTION and with selected .SECTIONs...

vhelin, 2023-04-28:
I kinda feel that WLA DX has become quite difficult to manage/learn/use as we have the default behavior and then there are plenty of ways to make it work differently, just to suit everybody's needs. Not sure what to do to this. Maybe everyone can live with it, or maybe there should be WLA DX 2, where the default behavior of many things have been changed to the most popular options. Or just add switches to change the default behaviors of this and that?

Kroc, 2023-04-28:
A lot of features have been added by creating new keywords and this rapidly increases the amount to remember to do certain things. There is perhaps not enough re-use of keywords or coherency between features but really the only difficult thing with using WLA DX I find is the sometimes confusing terminology. It seems WLA DX was made with ROM-based systems in mind first because variables are held in a RAM Section, yet when coding for the C64 the whole system uses RAM for both code and variables, and often these are intermixed inline instead of separated out. I would perhaps call these `VARSECTION`s. Likewise the use of "ROM banks" which are not ROM on RAM-banked systems like the ZX Spectrum 128.

vhelin, 2023-04-28:
True, WLA DX initially supported only systems that used ROM modules, but later support was added for different systems, and that was kind of a hack...

Kroc, 2023-05-06:
I'm going to close this since the heart of the problem was user error and you've provided both a work-around and a new feature that implements the work around. The question of naming things is one of the hardest problems in computer science and best left for another issue or discussion thread

closed 2023-05-06 by Kroc