vhelin/wla-dx #572: Refactor: Change as many g_* variables into s_*

vhelin, 2023-04-29:
There are plenty of variables that have g_* prefix (global), but are in fact used just inside that one file, so they could be turned into s_* (static) variables. One of these is g_dstruct_status in phase_1.c. There seems to be quite a lot of these so fixing them all will take quite a lot of time - even small commits will help in getting this done one day.

vhelin, 2023-05-25:
> So the way we will know when this is done is... there are no more
> variables we can rename without causing the compiler to error out?

Yeah, well, that’s one way of measuring it :) But even small commits where
this is being taken forward, help.

phase_1.c has lots of these and are usually directive parsing related.

lhsazevedo, 2023-06-22:
Here is a list of candidates I found:

WLA-CPU:
```
$ egrep -o "\<g_[A-Za-z0-9_]+" ./*.c | sort | uniq | cut -d":" -f2 | sort | uniq -u
g_after_tmp
g_bank
g_bank_defined
g_block_name_id
g_blocks
g_current_slot
g_data_stream_items_first
g_data_stream_items_last
g_defaultslot
g_defaultslot_defined
g_enumid
g_enumid_adder
g_enumid_defined
g_enumid_export
g_file_name_id
g_full_name_size
g_gbheader_defined
g_include_directives_name
g_instruction_n
g_instruction_p
g_label_sizeof_tmp
g_line_count_status
g_macros_last
g_nintendo_logo_dat
g_parse_dstruct_result
g_repeat_active
g_saved_structures_max
g_sizeof_g_error_message
g_skip_elifs
g_smsreservedspace_defined
g_snesemuvector_defined
g_snesheader_defined
g_snesnativevector_defined
g_stack_calculations
g_stack_calculations_max
g_stack_item_priority_items
g_stacks_inside
g_stacks_outside
g_unevaluated_expression
g_unknown_header_labels_last
g_unknown_labels_last
g_verbose_mode
g_version_string
g_wla_version
g_xyz
```

WLA-LINK:
```
$ egrep -o "\<g_[A-Za-z0-9_]+" ./wlalink/*.c | sort | uniq | cut -d":" -f2 | sort | uniq -u
g_object_files_array_max
g_object_files_max
g_section_table_table_array_max
g_si_operator_abs
g_si_operator_acos
g_si_operator_and
g_si_operator_asin
g_si_operator_atan
g_si_operator_atan2
g_si_operator_bank
g_si_operator_bank_byte
g_si_operator_ceil
g_si_operator_clamp
g_si_operator_compare_eq
g_si_operator_compare_gt
g_si_operator_compare_gte
g_si_operator_compare_lt
g_si_operator_compare_lte
g_si_operator_compare_neq
g_si_operator_cos
g_si_operator_cosh
g_si_operator_divide
g_si_operator_floor
g_si_operator_high_byte
g_si_operator_high_word
g_si_operator_log
g_si_operator_log10
g_si_operator_logical_and
g_si_operator_logical_or
g_si_operator_low_byte
g_si_operator_low_word
g_si_operator_max
g_si_operator_min
g_si_operator_minus
g_si_operator_modulo
g_si_operator_multiply
g_si_operator_negate
g_si_operator_not
g_si_operator_or
g_si_operator_plus
g_si_operator_pow
g_si_operator_power
g_si_operator_round
g_si_operator_shift_left
g_si_operator_shift_right
g_si_operator_sign
g_si_operator_sin
g_si_operator_sinh
g_si_operator_sqrt
g_si_operator_tan
g_si_operator_tanh
g_si_operator_unknown
g_si_operator_xor
g_stack_item_description
g_symbol_mode
```

vhelin, 2023-06-22:
I did not know you could do that! Just one line of commands in shell. Impressive! :) 

The WLALINK part should be easy as g_si_operator_* are just in main.c...

vhelin, 2023-06-24:
I took care of those variables in WLALINK, but forgot to mention this issue in the commit...

vhelin, 2023-07-17:
Now these are static (s_*) in WLA:

* g_after_tmp
* g_bank
* g_bank_defined
* g_block_name_id
* g_blocks
* g_current_slot
* g_data_stream_items_first
* g_data_stream_items_last
* g_defaultslot
* g_defaultslot_defined
* g_enumid
* g_enumid_adder
* g_enumid_defined
* g_enumid_export
+
* g_version_string
* g_wla_version

Also there is no g_xyz in WLA so that's taken care of as well. :) And g_verbose_mode was just an old extern that was unnecessary.

So what remains is (in WLA):

```

g_file_name_id
g_full_name_size
g_gbheader_defined
g_include_directives_name
g_instruction_n
g_instruction_p
g_label_sizeof_tmp
g_line_count_status
g_macros_last
g_nintendo_logo_dat
g_parse_dstruct_result
g_repeat_active
g_saved_structures_max
g_sizeof_g_error_message
g_skip_elifs
g_smsreservedspace_defined
g_snesemuvector_defined
g_snesheader_defined
g_snesnativevector_defined
g_stack_calculations
g_stack_calculations_max
g_stack_item_priority_items
g_stacks_inside
g_stacks_outside
g_unevaluated_expression
g_unknown_header_labels_last
g_unknown_labels_last

```

vhelin, 2023-07-17:
I left three *_last variables unchanged just to keep their *_first counterparts in the same reachability level. But all is done now. Thanks @lhsazevedo for the variable lists!

closed 2023-07-17 by vhelin