vhelin/wla-dx #577: Best way to deal with ROM mirroring (SNES)

rondnelson99, 2023-05-27:
I've used WLA a fair bit with Atari 2600 in the past, but I'm trying to use it for SNES now. I'm confused as to what the best way to approach mirroring and FastROM is. For anyone who isn't familiar, on SNES LoROM, banks $80 thru $FF are mostly a mirror of banks $0 thru $7F, but the SNES can be configured to access the higher bank number region at a faster speed. The root of the problem is this:

- The Header, vector table and vectors need to be in bank 0
- I want to use most other ROM addresses in banks $80-$FF
- I don't know how to approach this with `.MEMORYMAP` and `.ROMBANKMAP`.

I started out with a super simple memory map:
```.MEMORYMAP
	DEFAULTSLOT 0
	SLOT 0 START $8000 SIZE $8000 NAME "ROM"
	SLOT 1 START $0 SIZE $2000 NAME "LORAM"
	SLOT 2 START $FE2000 SIZE $1E000 NAME "WRAM"
.ENDME
```

Making sections in bank zero is straightforward this way, but I don't know how to define most other sections, which should have labels that are in bank $80+. I've tried simply using `.BANK $80`, but that doesn't work since I still want it to appear in the first bank of the ROM file. 

I've also tried putting everything in the bank $80 address range, with this:
```.MEMORYMAP
	DEFAULTSLOT 0
	SLOT 0 START $808000 SIZE $8000 NAME "ROM"
	SLOT 1 START $0 SIZE $2000 NAME "LORAM"
	SLOT 2 START $FE2000 SIZE $1E000 NAME "WRAM"
.ENDME
```
This didn't work either giving this error:
`obj/intro.o: src/intro.asm:18: _fprintf_snes_label(): The address of label "Intro" inside a bank is $808000 > $FFFF! Please submit a bug report!`
Do you have any advice here?

rondnelson99, 2023-05-27:
I've read some other forum posts and looked further into the documentation, and it seems like `.BASE $80` fits this use-case perfectly. So I guess I don't have any problem here. 

One potential suggestion: It seems like there are a lot of directives that are essentially just additional parameters for Sections, like `.ORGA`, `.BANK`, `.SLOT` etc, which can either be used standalone before a section, or in the section line as parameters (which makes a lot more sense IMO). `.BASE` seems like it would fit here in as well, as a paremeter you could pass on the section definition like to set it for that section. Should this be supported as well?

vhelin, 2023-05-27:
Good point about BASE missing from .SECTION (and also from .RAMSECTION) -> I just added it.

rondnelson99, 2023-05-27:
Awesome, thanks! I pulled the latest changes and it's working great.

closed 2023-05-27 by rondnelson99