vhelin/wla-dx #579: Improve: Remove g_input_parse_if and make A == B etc. conditions to produce a correct result, everywhere

vhelin, 2023-05-29:
The title says it all. Currently .IF forces the parser into a mode that can calculate A == B properly. If possible, make such calculations to work everywhere.

vhelin, 2023-05-30:
This includes making '<' (get low byte) etc. to work inside .IFs.

vhelin, 2023-06-10:
I've already completed this feature. Could not get totally rid of g_input_parse_if, but otherwise the feature seems to work, all of my tests pass.

Anyway, I'll push this to master right after v10.5 is released hopefully tomorrow...

closed 2023-06-23 by vhelin

maxim-zhao, 2023-06-28:
This does introduce some regressions in old code. In WLA DX, there's a bunch of places where we have lists of parameters with optional comma separators, and in general the parser will greedily consume things that look like expressions. Thus:
```asm
.db 1 2 3 4 5 ; 5 bytes
.db 1 -2 3 -4 5 ; 3 bytes!
.db 1, -2, 3, -4, 5 ; 5 bytes because we added commas
```
This change makes ">" and "<" able to act as binary operators in this context, so this code changes meaning
```asm
.db 1 <label 2 >label ; Used to be 4 bytes, now two (I think) - 1s or 0s depending on the label value
```
Having said that... I suspect it's better to fix up that old code rather than try to handle it gracefully.

vhelin, 2023-06-28:
> Having said that... I suspect it's better to fix up that old code rather than try to handle it gracefully.

I agree. Just thinking if there would be a way of helping the user to avoid such problematic situations... At least I can add a warning about this to the documentation.