vhelin/wla-dx #584: Label "_L" assembles unexpectedly

maxim-zhao, 2023-06-26:
In this code:

```asm
rticlesLower:        .dw _l,     _le, _la,     _les, _blank, _blank
ArticlesInitialUpper: .dw _L,     _Le, _La,     _Les, _blank, _blank
ArticlesPossessive:   .dw _de_l,  _du, _de_la,  _des, _d,     _de
ArticlesDirective:    .dw _a_l,   _au, _a_la,   _aux, _a,     _a
_l:     Article "'l"
_le:    Article " el"
_la:    Article " al"
_les:   Article " sel"
_blank: Article ""
_L:     Article "'L"
_Le:    Article " eL"
_La:    Article " aL"
_Les:   Article " seL"
_de_l:  Article "'l ed"
_du:    Article " ud"
_de_la: Article " al ed"
_des:   Article " sed"
_d:     Article "'d"
_de:    Article " ed"
_a_l:   Article "'l à"
_au:    Article " ua"
_a_la:  Article " al à"
_aux:   Article " xua"
_a:     Article " à"
```

...label `_L` assembles to `04 00`, which is not a valid value. In fact I can comment out the line defining that label and it compiles the same way.

While trying to isolate it to a minimal example, I came across `_b` and `_f` as "reserved" labels, but there's no warning/error if you try to define them. However I couldn't reproduce the weird behaviour on "_L" in this isolated test.

Are all `_[a-zA-z]` labels reserved? If so, please can you make it an error to try to redefine them?

maxim-zhao, 2023-06-26:
Some further digging finds it's my own fault. I have a `.define _L %0100` line elsewhere which is causing the confusion. However...

...shouldn't this be a compiler error? A label with the same name as a definition ought to be disallowed?

closed 2023-06-26 by maxim-zhao

vhelin, 2023-06-26:
> ...shouldn't this be a compiler error? A label with the same name as a definition ought to be disallowed?

This is a good point. I'll add checks to definition and label making plus disallow _b and _f definitions...

maxim-zhao, 2023-06-26:
I think uppercase _B and _F are also "reserved".