vhelin/wla-dx #586: incbin with filtermacro when binary data is not bytes

sverx, 2023-06-29:
This is not urgent at all, as I'm using a workaround that works fine anyway, but I wonder if there's a way to pass 16-bit words instead of bytes to the filter macro when using `.incbin`.

For instance, I have to add 0x1000 to each 16-bit value, which would be easier if I could do that directly instead of:

`.incbin "level_tilemap.bin" filter TILE_TRANSPARENCY`

```
.macro TILE_TRANSPARENCY
  ; this adds 0x1000 to each word
  .if (\2 & $01) != 0
    .redefine _out \1 + $10
   .else
    .redefine _out \1
  .endif 
.endm
```

Any idea? :smiley:

vhelin, 2023-06-29:
My only idea is to add support for something like

```
.incbin "level_tilemap.bin" filter TILE_TRANSPARENCY filtersize 2
```

... where "filtersize" would specify the size/number of bytes \1 in the filtermacro (and _out) specify... If this sounds like a plan I'll add it later this week.

vhelin, 2023-06-29:
Oh, and I think we might need something like "bigendian" and "littleendian" there as well...

sverx, 2023-06-30:
It might work!
Also, \2 would then be the position of that 'word' in the array, so it would still be 0,1,2,3... etc no matter the size of the data, right?
Also I think datasize should be limited to 1-4, at least for now.

sverx, 2023-06-30:
Also, how the macro itself would know how many bits the result have? Or it isn't important? :thinking:

vhelin, 2023-06-30:
I can keep \2 to be the position of the element, 0,1,2,3…

Yeah, 1-4 for the datasize. I don’t think we are ever going to add support
for bigger than 4 byte words as that would be a huge task, and I don’t know
any 8/16-bit architectures that support say 8 byte words.

vhelin, 2023-06-30:
The size of the result (_out) is the same as the input.

vhelin, 2023-06-30:
Ah, now I get you. We could supply the data’s size via \3.

vhelin, 2023-07-01:
> Oh, and I think we might need something like "bigendian" and "littleendian" there as well...

Actually .INCBIN already has SWAP which we can use to do the same thing...

vhelin, 2023-07-01:
Ok, does FILTERSIZE in .INCBIN work?

vhelin, 2023-07-01:
In your case you'd need to specify

.incbin "level_tilemap.bin" filter TILE_TRANSPARENCY filtersize 2

And then

.macro TILE_TRANSPARENCY
    .redefine _out \1 + $10
.endm

sverx, 2023-07-02:
Thanks. I will try that and give you feedback. :heart: 

Regarding `SWAP`, I believe that its function should remain as it is now, something that swaps the order of odd/even **bytes** (so nothing to do with filters) and you might want to provide a `BIGENDIAN` option **for filters** in case the provided data (when filtersize is 2 or more) has to be treated as so. That's something I personally don't need but it might be an interesting addition nonetheless.

sverx, 2023-07-02:
Oh I see now checking your code that you extended `SWAP` to work on bigger chunks than a single byte too!
That's indeed interesting, and it doesn't seem to break compatibility so... great job! :+1:

sverx, 2023-07-03:
It seems to be working correctly, at least in my use case.
Thanks! :heart:

closed 2023-07-03 by vhelin