vhelin/wla-dx #590: eZ80 support -- AgonLight

Kroc, 2023-07-07:
The [AgonLight](https://www.thebyteattic.com/p/agon.html) is a cheap (€50) single-board microcomputer available from multiple retailers across the globe (US, EU, China). It uses an [eZ80](https://www.zilog.com/docs/um0077.pdf), a modern, single-cycle Z80 with an optional 24-bit addressing mode somewhat like the 65C816.

Currently, software is developed using Zilog's clunky, Windows-only ZDS II IDE, at least on PC (there is a [native assembler](https://github.com/envenomator/agon-ez80asm) for those looking to develop on the device itself). Looking into how software is developed for the platform, I see a great opportunity for WLA-DX to be the standard assembler for AgonLight and to alleviate the community of their Windows-only dependency.

You should speak to @TheByteAttic and @breakintoprogram who developed the platform and software and could explain the CPU and their needs more precisely.

vhelin, 2023-07-08:
Sounds like a plan!

@TheByteAttic @breakintoprogram or someone else - I checked the eZ80 manual, but it didn't have a list/table of changes versus Z80 - it seems eZ80 instructions are a superset of Z80 instructions so if someone could give a link/pointer to what I need to add to add eZ80 compability - that'd be great!

vhelin, 2023-07-16:
After looking at the eZ80 documentation I can see we have ADL mode there, which affects quite a lot. My next question is:
Are Z80's undocumented/documented opcodes supported on eZ80 i.e., should I copy Z80 support in WLA DX as a starting point for eZ80 support or start from zero?

vhelin, 2023-07-16:
Also, do we need some kind of a program header for AgonLight?

Kroc, 2023-07-16:
Not externally to the in-memory binary, no. Agon MOS programs just need some meta-data after the first 3 bytes, which are a jump to the code. https://github.com/breakintoprogram/agon-projects/blob/main/ASM/Hello%20World%2024/init.asm

vhelin, 2023-07-16:
So the programmer can just define one 16MB "ROM bank" and populate that meta-data area starting at byte 64 and then write everything out to a file starting at byte 0 and ending at the last touched byte?

Kroc, 2023-07-16:
Yes, this appears so, at least as far as building the sample code has done for me.