vhelin/wla-dx #593: Tmpfile on Windows will often not work, cascades into `PARSE_STACK` error on fix.

cr1901, 2023-07-20:
Starting from ee3329c8c57, we use `tmpfile` to create temporary files. When running tests, we will get output similar to this on Windows, where I added a `perror` call:

```sh
William@DESKTOP-3H1DSBV MINGW64 ~/Projects/legacy/wla-dx
$ ./run_tests.sh
Building byte_tester...
make ; cp byte_tester ../binaries
make[1]: Entering directory '/home/William/Projects/legacy/wla-dx/byte_tester'
gcc  main.o -o byte_tester
make[1]: Leaving directory '/home/William/Projects/legacy/wla-dx/byte_tester'

########################################################################
WARNING: Valgrind is not installed so we cannot perform memory checks...
########################################################################

Running tests...

rm -f main.o more.o core *~ *.lst linked.rom linked.sym
wla-6502 -i -o main.o main.s
Permission denied
MAIN: Error creating a tmp file for WLA's internal data stream.
make: *** [makefile:15: main.o] Error 1

########
FAILURE!
########
Test "after_test/" of platform "6502/" failed.
```

`perror` returns `Permission denied`. It turns out that for historical reasons, Windows writes tmpfiles to the root directory, which you often don't have permission to write to. [Microsoft docs](https://learn.microsoft.com/en-us/cpp/c-runtime-library/reference/tmpfile?view=msvc-170#remarks) recommends using `tmpnam` and `fopen`. I've supplied a _partial_ patch here:

```diff
diff --git a/main.c b/main.c
index 782e99a..33348ee 100644
--- a/main.c
+++ b/main.c
@@ -89,6 +89,10 @@ static int s_deletable_structures_max = 0, s_deletable_structures_count = 0;
 extern char *g_sdsctag_name_str, *g_sdsctag_notes_str, *g_sdsctag_author_str;
 #endif
 
+#ifdef WIN32
+  static char * s_tmpfile_name;
+#endif
+
 PROFILE_GLOBALS();
 
 
@@ -526,6 +530,11 @@ static void _procedures_at_exit(void) {
   if (g_file_out_ptr != NULL) {
     fclose(g_file_out_ptr);
     g_file_out_ptr = NULL;
+
+#ifdef WIN32
+    remove(s_tmpfile_name);
+    s_tmpfile_name = NULL;
+#endif
   }
 
   free(g_macro_stack);
@@ -951,7 +960,28 @@ int main(int argc, char *argv[]) {
     return 1;
   }
 
+#ifdef WIN32
+  /* tmpfile uses root of drive on Windows, which is probably not-writable. */
+  s_tmpfile_name = tmpnam(NULL);
+
+  if (s_tmpfile_name == NULL) {
+    fprintf(stderr, "MAIN: Error creating a tmp filename for WLA's internal data stream.\n");
+    return 1;
+  }
+
+  /* Per: https://learn.microsoft.com/en-us/cpp/c-runtime-library/reference/tempnam-wtempnam-tmpnam-wtmpnam?view=msvc-170&redirectedfrom=MSDN
+  When a file name is prepended with a backslash and no path information,
+  such as \fname21, it indicates that the name is valid for the current working
+  directory. */
+  if(s_tmpfile_name[0] == '\\') {
+    s_tmpfile_name = s_tmpfile_name + 1;
+  }
+
+  g_file_out_ptr = fopen(s_tmpfile_name, "wb");
+  fprintf(stderr, "%s\n", s_tmpfile_name);
+#else
   g_file_out_ptr = tmpfile();
+#endif
   if (g_file_out_ptr == NULL) {
     fprintf(stderr, "MAIN: Error creating a tmp file for WLA's internal data stream.\n");
     return 1;

```

Unfortunately, after applying this patch, the link fails with a `PARSE_STACK` error:

```
$ ./run_tests.sh
Building byte_tester...
make ; cp byte_tester ../binaries
make[1]: Entering directory '/home/William/Projects/legacy/wla-dx/byte_tester'
gcc  main.o -o byte_tester
make[1]: Leaving directory '/home/William/Projects/legacy/wla-dx/byte_tester'

########################################################################
WARNING: Valgrind is not installed so we cannot perform memory checks...
########################################################################

Running tests...

rm -f main.o more.o core *~ *.lst linked.rom linked.sym
wla-6502 -i -o main.o main.s
shm0.
wla-6502 -i -o more.o more.s
s754.
wlalink -v -i -s linkfile linked.rom
more.o: more.s:78: PARSE_STACK: Unresolved reference to "label_2X".
make: *** [makefile:12: all] Error 1

########
FAILURE!
########
Test "after_test/" of platform "6502/" failed.
```

Have any idea on the follow-up error? I've attached my [after_test directory](https://github.com/vhelin/wla-dx/files/12102179/after_test.zip) with assembler output. Patch should be applied against: 45fc096f90b4107e2bdbfcc4f0d07130beac7e7a.

vhelin, 2023-07-20:
I'm using Cygwin under Windows 11 myself to develop WLA DX and there everything works as normally. But I see you are using MinGW64? I'll try to install that and recreate the problem...

vhelin, 2023-07-20:
Ok, it seems I don't know how to use MinGW

[image]

What is this shell? I think I need more info about how to set up the same environment that you have...

[image]

vhelin, 2023-07-20:
I ran your after_test under Cygwin:

[image]

First without assembling anything, just linking -> got the same error as you got. After make clean it worked just fine. I assume your version of wla-6502 is somehow broken...

vhelin, 2023-07-20:
I added your patch with small mods, but now we have your PARSE_STACK issue...

vhelin, 2023-07-20:
I think it was this. I changed it to "w+" as we are also reading from that file and Windows x86 Azure Pipelines task completed successfully.

[image]

If you try out the latest sources, does it work now?

cr1901, 2023-07-20:
Sorry, I was asleep. You were busy while I was asleep :D!

>What is this shell? I think I need more info about how to set up the same environment that you have...

I'm using [MSYS2](https://www.msys2.org/) to get a `bash` shell _and_ the `mingw-w64` compilers. Looks like you've installed just the `mingw-w64` compilers.

>If you try out the latest sources, does it work now?

Ugh, the documentation _even says_ [tempfiles use](https://learn.microsoft.com/en-us/cpp/c-runtime-library/reference/tmpfile?view=msvc-170#remarks) `+`!!

> The temporary file is opened in w+b (binary read/write) mode.

I thought it meant `w` _and_ `b`, and forgot the `+` can also be in the middle. I added the `b` locally as well to be extra safe.

Now we get further into the tests, but now we fail at one of the 8080 tests:

```
./run_tests.sh
[30/30] Linking C executable binaries\wla-huc6280.exe
Building byte_tester...
make ; cp byte_tester ../binaries
make[1]: Entering directory '/home/William/Projects/legacy/wla-dx/byte_tester'
gcc  main.o -o byte_tester
make[1]: Leaving directory '/home/William/Projects/legacy/wla-dx/byte_tester'

########################################################################
WARNING: Valgrind is not installed so we cannot perform memory checks...
########################################################################

Running tests...
.............................................................

rm -f core *~ linked.rom linked.sym *.o
wla-8080 -M -o main.o main.s
main.s:138: INCLUDE_FILE: Error creating a tmp file for "includeXYZ3.s"!
main.s:138: ERROR: Couldn't parse ".include".
make: *** [makefile:11: main.o] Error 1

########
FAILURE!
########
Test "makefile_generation/" of platform "8080/" failed.
```

Hmm...

vhelin, 2023-07-20:
If you rerun the tests do you still get the same error in the same place? I'm wondering if that is now something that happens randomly once in a while as the Windows x86 Azure Pipelines task completed successfully...

vhelin, 2023-07-20:
Could you debug why you get this?

[image]

Like fprintf() the tmp fle name or step it in a debugger?

vhelin, 2023-07-20:
I ran the x86 Azure Pipelines task

[image]

two times again, manually, and both times it succeeded.

vhelin, 2023-07-20:
Ok, I installed MSYS2 and all tests passed. Perhaps I'm using it differently than you?

[image]

cr1901, 2023-07-20:
Okay something is clearly weird on my end then. I'll debug now.

cr1901, 2023-07-20:
I think I found the culprit: https://github.com/vhelin/wla-dx/blob/master/include.c#L140

There's other `tmpfile` calls in the source. Perhaps spin out to a `tmpfile.c` source file? I can try to take a stab at this.

Also:

Should [this line](https://github.com/vhelin/wla-dx/blob/master/tests/8080/makefile_generation/main.s#L137) be succeeding in tests?

```
.include { "include{XYZ_STR}3.s" }
```

The source file in  `tests/8080/makefile_generation` is called `includeXYZ.s`, not `includeXYZ3.s`.

vhelin, 2023-07-20:
It seems tmpfile() is called in three .c files! I had totally forgotten this...

[image]

Making tmpfile.c would require changes to Visual Studio Solution and makefiles etc., perhaps just make the tmpfile etc. creation function public and place it in main.c?

vhelin, 2023-07-20:
Well, if you want to update all the Visual Studio Solutions and historical makefiles via editing the makefile generator etc., feel free to add a new .c file to the project. :)

commit d7f20ec5 (2023-07-20): Remove all tmpfile calls on Windows due to poor legacy behavior. Fixes #593.
commit 02286745 (2023-07-20): Remove all tmpfile calls on Windows due to poor legacy behavior. Fixes #593.
PR #595 (2023-07-21): Remove all tmpfile calls on Windows due to poor legacy behavior. Fixe…
Fixes #593.

Use `extern int create_tmp_file(FILE **);` to avoid the need for a new source file for now. We may wish to split it out in the future.
closed 2023-07-21 by vhelin