vhelin/wla-dx #602: Override address for SMSHEADER

lajohnston, 2023-08-19:
Hi there,

The Master System supports 3 addresses to place the ROM header and checksum ($1FF0, $3FF0, $7FF0, see docs [here](https://www.smspower.org/Development/ROMHeader)). It seems this is currently based on the highest available location according to the ROM size (8KB, 16KB, 32KB+ respectively). Would it be possible to add an ADDRESS or LOCATION option to the SMSHEADER directive to override the default address to one of these values? If so, would the address be respected by the other SMS directives such as SMSTAG, COMPUTESMSCHECKSUM, and also SDSCTAG (which also seems to place the header as well)?

My use case is a bit niche, but I'm finding the default location ($7FF0) breaks up the contiguous free ROM space and is currently getting in the way.

vhelin, 2023-08-19:
I could try to add something like hmm BASEADDRESS (?) tomorrow. Need to change the object/library file format for this, but it's not a big deal.

lajohnston, 2023-08-19:
Great, thanks! 🙂

vhelin, 2023-08-20:
>  If so, would the address be respected by the other SMS directives such as SMSTAG, COMPUTESMSCHECKSUM, and also SDSCTAG (which also seems to place the header as well)?

Looking at SDSCTAG specification it's always written at 0x7FE0, so BASEADDRESS doesn't affect that. Or should it be BASEADDRESS - $10?

lajohnston, 2023-08-20:
Thanks, according to https://www.smspower.org/Development/SDSCHeader that should always be at $7fe0. It might be best to fix that there for compatibility with the software that reads it:

> The SDSC tag is found at offset $7fe0, directly before the standard [ROM header](https://www.smspower.org/Development/ROMHeader) (if it's in the usual place).

In my case I'll just have to leave that out of my ROM so it doesn't get in the way

vhelin, 2023-08-20:
I made SDSC tag to be always placed 16 bytes below the ROM header as that was the old behaviour in WLA DX - there might be a reason why it worked like that...

lajohnston, 2023-08-20:
thanks, that's working fine for me :) It looks like it currently allows addresses other than $1FF0, $3FF0, $7FF0 which are the only valid ones, but I don't think that's a too much of a problem if it's mentioned in the docs and used correctly.

The current -$10 offset for the SDSC tag is probably sensible, as for 8KB and 16KB ROMs $7fe0 would be out of bounds. If anything maybe the SDSC spec + software could be updated to support those locations, but from WLA DX's perspective I think it's better better for it to include the tag at the 'wrong' location than leave it out completely 👍

closed 2023-08-20 by lajohnston

maxim-zhao, 2023-08-20:
I'll go change the SDSC spec to match :)