vhelin/wla-dx #604: [BUILD REQUEST] 64bit Windows Build is needed!!!

TheRealHamtaro126, 2023-08-22:
I can't have 32-bit anymore, I now have a Win11 computer... Without 64bit support I cannot do anything with this...

This is the only limit that no-one encountered, but me... Hope you can do something!

God bless the 6502/65816 Assembly communities for the hard work!!!

vhelin, 2023-08-22:
As a quick fix I could try to build x64 of v10.5 manually, or do you need
the latest master?

Hmm could we automate this somehow with Azure?

TheRealHamtaro126, 2023-08-22:
Manually, Please...

And thanks for the big help!!!

TheRealHamtaro126, 2023-08-22:
I would love to try to stick to the v10.5 build for now...

Good luck!!!

vhelin, 2023-08-22:
LOL, my memory is really bad; we already have x64 building automatically in Azure:

[image]

Here you should be able to get the build atrifacts: https://dev.azure.com/villehelin0486/a02dde9d-6d39-4906-a9ab-3e8937b201bf/_apis/build/builds/2910/artifacts?artifactName=drop&api-version=7.0&%24format=zip - does this work for you? It's the latest master, but if you really need v10.5 I can do that, too...

vhelin, 2023-08-22:
I build x64 of v10.5 and added it here: https://github.com/vhelin/wla-dx/releases/tag/v10.5

In the future I'll make x64 Windows releases as well as x86...

TheRealHamtaro126, 2023-08-22:
Thank you very much... #604 should be solved...

I will report back once I find anything... May also Keep this open for anything...

TheRealHamtaro126, 2023-08-22:
Glitches, Bugs, Y'name it... I'll be there...

vhelin, 2023-08-23:
Let's close this issue as we now have 64-bit x64 release (v10.5 and more in the future). If you spot any bugs in the x64 builds, please open new issues, one per issue.