vhelin/wla-dx #606: Segmentation fault when

Jellonator, 2023-08-26:
I have created a minimal example for this crash.
This example is for the 65c816, and I am compiling for the SNES.
I am using wla-dx v10.6a

```asm
.MEMORYMAP
    DEFAULTSLOT 0
    SLOT 0 $8000 $8000 "ROM" ; ROM slot
.ENDME

.ROMBANKMAP
    BANKSTOTAL $80
    BANKSIZE $8000
    BANKS $80
.ENDRO
.BASE $80
.SNESHEADER
    ID "SNES"
    NAME "Test"
    FASTROM
    LOROM
    CARTRIDGETYPE $00
    SRAMSIZE $00
    COUNTRY $01
    LICENSEECODE $00
    VERSION $00
.ENDSNES

.EMPTYFILL $00

; actual example
.BANK $00 SLOT "ROM"
.SECTION "TEST" FREE

.DEFINE WIDTH 10
lda loword(mapDoorVertical)-WIDTH
lda loword(mapDoorVertical)

.ENDS
```

When I run `wla-65816 -I include -o tmp.obj tmp.asm`, I get a segmentation fault (core dumped):
```
$ wla-65816 -v -I include -o obj/tmp.obj tmp.asm
Pass 1...
Segmentation fault (core dumped)
```

No other information is given by the assembler.
In my testing, this seems to be the relevant info:
* this is not specific to LDA, and occurs with any access mode
* this only occurs if I have both instructions.
* The instruction with the `-WIDTH` must be first.
* Any number of instructions may precede, be between, or come after them.
* If I move the `-WIDTH` into the `loword(...)` call, then it assembles correctly.

vhelin, 2023-08-26:
Thanks! We rarely get segfaults, but all of them need to be fixed. :) I'll look into this tomorrow.

vhelin, 2023-08-27:
In this case the calculation stack would initially be as follows:

1. loword()
2. mapDoorVertical
3. - (minus)
4. WIDTH

WLA would incorrectly think that there's an opportunity to calculate A-B in 2-4... I marked now function arguments so that they cannot be used in A-B calculations (but A-B works inside one argument, A-B just doesn't overflow from the arguments any more).

vhelin, 2023-09-02:
@Jellonator Do the latest sources fix this issue for you? At least my tests pass now...

vhelin, 2023-10-31:
I assume my fix solved this issue. If not, please reopen this issue.

closed 2023-10-31 by vhelin