vhelin/wla-dx #607: .stringmaptable file not respecting incdir

lajohnston, 2023-08-28:
Hi there,

I'm using `.incdir` in a project to point to a different directory and all the `.include`s are working fine, but when passing a file to .stringmaptable it seems it's still pointing to the original file directory, i.e.

File structure
```
/someDir
   foo.asm
   stringmap.tbl
main.asm
```

main.asm
```
.incdir "./someDir"
.include "foo.asm" ; works
.stringmaptable script "stringmap.tbl" ; Error opening file
```

Although .incdir is currently used for `.include` and `.incbin`, is it also worth extended it to other files paths or is there another way to do this?

vhelin, 2023-08-28:
The funny thing is that .STRINGMAPTABLE had code that tried to use the directory specified by .INCDIR, but it was buggy. :)

lajohnston, 2023-08-28:
Thanks, that seems to fix it 🙂

closed 2023-08-28 by lajohnston

maxim-zhao, 2023-08-29:
Hmm, sorry about that :) I think I based it on `.incbin` or something similar but then it got a bit confused along the way, and I apparently never used this feature. I'm happy to add advice on how I intended to use it. 

By the way @vhelin this feature exceeds `.asciitable` by mapping one or more string bytes (usually characters, maybe multi-byte characters) to one or more bytes - you can specify any even number of hex characters before the `=`. So your documentation change is a little bit wrong.

vhelin, 2023-08-29:
No worries! It's better to have contributions with occasional bugs here and there instead of having no contributions at all. Bugs happen to people who write code. :)

I have to say about the document change that I thought that old documentation about .STRINGMAPTABLE was copied from somewhere else (.ASC?) and left without updates. Or it didn't make any sense to me. But then again I had a couple of beers yesterday evening... :) Anyway, I'll update the documentation to reflect the fact that .STRINGMAPTABLE can be used to map a single char to multiple bytes...

maxim-zhao, 2023-08-29:
Yes, the documentation is mostly under `.stringmaptable`. 

One side note is that the example there is reportedly no longer valid due to the way WLA now interprets backslashes in strings - so the table has "\n" as a two byte string and the string to be mapped has it converted to the control code before the directive code sees it. I'm not sure if there's an easy way to fix that?

vhelin, 2023-08-29:
Hmm maybe the .STRINGMAPTABLE entries' texts should be processed the same way .STRINGMAP's input string is processed in input_number()? process_string_for_special_characters() (in parse.c)?