vhelin/wla-dx #608: Problems with labels arithmetic

leafi, 2023-09-08:
Hi, I've been having some trouble figuring out how label arithmetic should work.

I have a dynamic dispatch thingy in GBZ80 code where I'm trying to ensure the length of a section doesn't grow longer than 256 bytes.

I've tried both the 10.5 release and the latest master as of today.

I'm trying to use either `.assert` or `.if` to check the upper bytes of label addresses match. Here's some sample code; ignore the memorymap & rombankmap:

```
.memorymap
    defaultslot 0
    slot 0 start $0000 size $4000 name "ROM0"    ; mbc5 fixes to first 16k rom bank
.endme
.rombankmap
    bankstotal 4  ; 64k for now
    banksize $4000  ; 16k each
    banks 4       ; for now, all banks are standard size
.endro

.bank 0 slot 0
.orga $100
.section "code.testo" semifree keep align 256
label1:
    push af
    ret
label2:
    push hl
    ret
@testo_end:
.ends

.if label1 == label2
    .fail "just testing whether this .if can parse"
.endif

.assert >label1 == >label2@testo_end

.assert (hibyte(label1) == hibyte(label2@testo_end))

```

I can't seem to get the .if or .assert to parse at all:

```
[leaf@alicorn zorro]$ wla-gb 2>&1 | grep v10
---               WLA GB-Z80 Macro Assembler v10.6a                ---

[leaf@alicorn zorro]$ wla-gb -v -o test.o src/test.gb.asm
Pass 1...
src/test.gb.asm:23: INPUT_ERROR: .IF needs immediate data.
src/test.gb.asm:23: ERROR: Couldn't parse ".if".
```

Commenting out line 23 & 24 & 25 (the .if ... .endif):

```
[leaf@alicorn zorro]$ wla-gb -v -o test.o src/test.gb.asm
Pass 1...
src/test.gb.asm:27: INPUT_ERROR: .ASSERT needs immediate data.
src/test.gb.asm:27: ERROR: Couldn't parse ".assert".
```

...and, with line 27 gone, the same for line 29 (the 'hibyte()' line).

Any ideas? Thanks for your efforts.

vhelin, 2023-09-08:
> .assert >label1 == >***@***.***_end
> .assert (hibyte(label1) == ***@***.***_end))

The assembler is currently only able to calculate A-B for labels. It should
be able to know the address of a label outside of a .SECTION, but there no
code to do that so you cannot give a plain label to a directive that
expects immediate data.

I could try this weekend to fix this issue, to make the assembler smarter...

leafi, 2023-09-08:
Ahh, it's only A-B that works for labels -- thanks, this helps a lot!

Forgive the big block of code, but please consider this:

```
.memorymap
    defaultslot 0
    slot 0 start $0000 size $4000 name "ROM0"    ; mbc5 fixes to first 16k rom bank
    slot 1 start $c000 size $1000 name "WRAM0"
.endme
.rombankmap
    bankstotal 4  ; 64k for now
    banksize $4000  ; 16k each
    banks 4       ; for now, all banks are standard size
.endro

.ramsection "wram0.test" slot "WRAM0" orga $c000 force
    var1   db
    .      dsb 254
    var2   db
    var3   db

;.assert (var2 - var1) >> 8 == 0   ; <- Unexpected symbol "(var2" in .RAMS
.ends

;.assert >var2 == >var1             ; <- E: Need immediate
.assert (var2 - var1) >> 8 == 0
.assert (var3 - var1) >> 8 == 1

.bank 0 slot 0
.orga $100
.section "code.testo" semifree keep align 256
label1:
    push af
    ret
label2:
    push hl
    ret
@testo_end:
.assert (@testo_end - label1) >> 8 == 0
;.assert >@testo_end == >label1              ; <- E: Need immediate
;.assert (@testo_end >> 8) == (label1 >> 8)  ; <- E: Need immediate

.ends

.assert (label2@testo_end - label1) >> 8 == 0

```

(All uncommented asserts work here.)

1. At least here, it doesn't seem to matter whether .assert is inside a .section or not -- **except** that .assert doesn't work in .ramsection ("src/test.gb.asm:19: DIRECTIVE_ERROR: Unexpected symbol "(var2" in .RAMSECTION.")
2. Am I making a mistake by putting vars in .ramsections? I've been doing it to try and keep size info & allow the linker to know which mem is carved out, but should I be .define'ing vars to mem addrs instead? It's seeming like label arithmetic is a niche feature which I really shouldn't be doing. Arithmetic is meant for defines, right?

I don't want you to add features purely on my behalf, particularly if I'm using the assembler in the completely wrong way. This is the first time I've seriously written assembly.

leafi, 2023-09-08:
OK, I've been thinking things through... So the assembler can't really know where labels are until linking is done, I guess. "Need immediate" meaning "labels aren't immediately placed", rather than "you must type a literal number here".

I guess, then, I'm surprised even A-B works at all. My generalized .assert style here would require the assembler to collect asserts and defer them to link time, which is absurd. I think I should remove the asserts and use the 'size NNN' constraints on .section / .ramsection lines.

Sorry for the spam, and thanks for your quick reply. When you get a moment, please say whether my thinking is right, and tell me which (if any) behavior you want broken out to a separate issue with a test case, or close the issue if there's nothing more to do here. No rush.

vhelin, 2023-09-08:
About this

```
.ramsection "wram0.test" slot "WRAM0" orga $c000 force
    var1   db
    .      dsb 254
    var2   db
    var3   db

;.assert (var2 - var1) >> 8 == 0   ; <- Unexpected symbol "(var2" in .RAMS
.ends
```

.RAMSECTION is not really a normal .SECTION, it's more a .STRUCT like directive with its own parser. If you move that .ASSERT after .ENDS then it should work

vhelin, 2023-09-08:
About these

```
;.assert >@testo_end == >label1              ; <- E: Need immediate
;.assert (@testo_end >> 8) == (label1 >> 8)  ; <- E: Need immediate
```

These won't work as the assembler specifically, at the moment, looks for A-B calculations and there tries to find out if it can calculate the delta. The A-B feature was implemented, because people often wanted to feed the size of a code block (between labels A and B) to some directives that needed immediate data.

vhelin, 2023-09-08:
> 2. Am I making a mistake by putting vars in .ramsections? I've been doing it to try and keep size info & allow the linker to know which mem is carved out, but should I be .define'ing vars to mem [...]

I think it's up to you if you use .RAMSECTIONs that the linker can place whereever it finds space and according to the type of the .RAMSECTION of use .DEFINEs.

Label arithmetic should work in normal calculations, but not with many directives as the assembler parses and implements those directives then it needs to know the result of the calculation while assembling. But if you make a calculation that is postponed to the linker, like

.db A-B

or

   lda A-B

then there is no problem, the linker can solve these.

> I don't want you to add features purely on my behalf, particularly if I'm using the assembler in the completely wrong way. This is the first time I've seriously written assembly.

This is a good opportunity to get some feedback from "fresh eyes". :)

vhelin, 2023-09-08:
> I don't want you to add features purely on my behalf, particularly if I'm using the assembler in the completely wrong way. This is the first time I've seriously written assembly.

I think the need to make the assembler to actually know the address of labels outside .SECTIONs is a pretty good feature and now that we have the capability to calculate A-B in the assembler, perhaps I can modify that code to find addresses of labels outside .SECTIONs, not just deltas.

vhelin, 2023-09-10:
It wasn't quite trivial, but I managed to make the assembler to turn labels outside .SECTIONs (and inside FORCE and OVERWRITE .SECTIONs) into address values as long as they are defined before referenced. I made a branch about this:

https://github.com/vhelin/wla-dx/tree/turn_labels_into_address_values

I need to improve the tests before I'm confident I didn't break anything...

vhelin, 2023-10-01:
I merged this branch to master, please report any issues you see!

vhelin, 2023-10-31:
I take that this issue has been solved by the commit I made. If you still have such troubles reopen this issue or open a new one.

closed 2023-10-31 by vhelin