vhelin/wla-dx #609: Appending sections: 'No room for section'

lajohnston, 2023-09-11:
Hi there,

Are there any known issues with adding a section `after` another section, when nothing else has been appended to it with `appendto`? I have a project the contains the following code, which errors with `INSERT_SECTIONS: No room for section "footer" (1 bytes) in ROM bank 0`, despite there being plenty of room in bank 0.

```asm
.section "header" free keep
    header:
        nop
.ends

.section "footer" after header keep
    ret
.ends
```

The reason I don't have anything that appends to this with `appendto` is that it's an optional user-specified hook in a library, so it may or may not exist. Adding the following dummy one _appears_ to fix it:

```asm
.section "body" appendto header
    nop     ; some code
.ends
```

I can do that as a workaround, but just interested if I'm doing anything wrong. It seems to be erroring like this at other random times, and I can't work out what's going wrong or how to replicate it other than in this scenario above, so I don't know whether there's anything flaky in my code. I've been doing something similar with `ramsections` but haven't encountered the issue there.

As a side note, is there any way I can keep the 'header' label without putting the dummy `nop` instruction after it? It's not too much of a problem as it's only a few cycles, but just interested :)

vhelin, 2023-09-11:
My brain tells me something like this has happened in the past, but not sure what.

Anyway, I tried to assemble

```
.section "header" free keep
    header:
        nop
.ends

.section "footer" after header keep
    ret
.ends
```

... and it works just fine.

My initial suspicion is that you are mixing AFTER and APPENDTO .SECTIONs. Maybe just use AFTER or APPENDTO .SECTIONs, not both in this case?

vhelin, 2023-09-11:
> As a side note, is there any way I can keep the 'header' label without putting the dummy nop instruction after it? It's not too much of a problem as it's only a few cycles, but just interested :)

So you mean you'd like to have empty .SECTIONs? Is that the case? We could make empty (no real data/bytes there) .SECTIONs to work...

lajohnston, 2023-09-12:
> My initial suspicion is that you are mixing AFTER and APPENDTO .SECTIONs. Maybe just use AFTER or APPENDTO .SECTIONs, not both in this case?

Yes, the use-case for mixing them is that it allows zero or more sections to be appended to a section, then allows one thing to get added at the end to cap it off. In this case it's a `ret`, without which the code block would become invalid. Reading the docs I thought that was the use-case for `after`  that distinguishes it from `appendto`, otherwise if they can't be mixed it seems like they do the same thing.

I've unfortunately not been able to reproduce it in a simple example, and it appears and disappears when I comment out random pieces of code, presumably because it shuffles the sections around. The closest I've got to one is in this file/branch: https://github.com/lajohnston/zest/blob/wladx-issue-609/examples/01-basic/main.asm

This example has fixed code blocks as well and in this case perhaps it's trying to allocate the section in between two of these fixed blocks and there's not enough room. Usually a free section would find somewhere else to go but it seems this one with `appendto` and `after` stuff isn't. Does WLA-DX calculate the full size of the section with appendto and after before calculating where to place it?

In the example there are two dynamic sections, one in `preSuite.asm` and one in `preTest.asm`. Commenting out one or the other seems to fix it and it then errors at a missing label (which is expected in this butchered code). Commenting out a line of code in the fixed $0000 code block (`ld sp, $dff0`) in main.asm also fixes it, which is what makes me think it's trying to allocate it around there.

> So you mean you'd like to have empty .SECTIONs?

Yes, in this case the the header section would be empty in itself, but because stuff gets appended to it will have a size in the final build.

vhelin, 2023-09-12:
I have to admit that AFTER and APPENDTO .SECTIONs are quite similar, but I'm thinking that you can APPENDTO .SECTIONs together and then have that single .SECTION be treated as a big chunk. AFTER .SECTIONs can have distances between them so there you can have in a way aligned .SECTIONs, like first being at x, next one at x+$100, next on at x+$200...

If you have just APPENDTO .SECTIONs then the linker should find a suitable place for that resulting big .SECTION, but in the case of AFTER .SECTIONs the first one gets its place and then the rest that come after are placed in the output according to their distances. If there is already something there then tough luck.

Not 100% sure about this, though, it's been a while since I last touched this stuff, but I don't think the .SECTION placer takes into account the space the AFTER .SECTIONs occupy when it's placing a series of them, just the first one... You might try giving .SECTIONs priorities that are linked in an AFTER chain, perhaps that will help?

When using APPENDTO the .SECTION merger should first merge the .SECTIONs and then find room for the big combined .SECTION and it should know the full size of the merged .SECTION.

I have a deadline at work next week so I don't know if I can investigate this during this week. I see you can give -k to the assembler so it'll keep empty .SECTIONs.

lajohnston, 2023-09-13:
> I don't think the .SECTION placer takes into account the space the AFTER .SECTIONs occupy when it's placing a series of them

Ok thanks, that sounds like it's crux of the issue in my case. I just have one `after` section but it seems that then clashes with fixed code. Worst comes to worst I could try forcing the start of the section outside that fixed code region now I know that's the issue.

> You might try giving .SECTIONs priorities that are linked in an AFTER chain, perhaps that will help?

I've just tried that with `appendto`, using `.section "footer" appendto mySection priority 65535`, but it seems these priorities are being ignored unless I'm doing something wrong. Would appendto respect those priority values for these appending section or is the linker just taking the parent section's priority? I prefer the terminology of `appendto` in my use-case but if that's the wrong thing to be using I can just switch to using `after` :)

vhelin, 2023-09-13:
Try giving those AFTER .SECTIONs also that same high priority. Sorry, I'm busy with work, deadline next Tuesday and I'm still doing work related tasks...

lajohnston, 2023-09-13:
That's ok, there's no rush for this :) I'll try out some things

vhelin, 2023-09-13:
I you give AFTER .SECTIONs the same high priority than the .SECTION they come after, and it works, I'm thinking that the AFTER .SECTIONs should perhaps inherit the priority from the .SECTIONs they follow. No idea if that is implemented, perhaps not, it's 02:35AM here and I just finished working, but yeah, let's come back to this later. :)

lajohnston, 2023-09-20:
Ok, so weird thing is that using `appendto` seems to actually work for this, contrary to my expectations. If I have 2 sections (header and footer) in a file, one of which uses `appendto` to get appended to the other one, when another file that is included elsewhere appends to the section, that actually gets appended first before the original. The footer therefore remains in last place as I'd want it (but wouldn't expect).

An example is [here](https://github.com/lajohnston/zest/blob/main/examples/09-hooks/suite.asm), where after I `.include` the [original files](https://github.com/lajohnston/zest/blob/main/src/preSuite.asm) containing the header and footers, then append some more stuff to the header.

I thought it might be operating on a last-in-first-out basis, but then having other `appendto` sections elsewhere seem to get added in the order they're specified. I'm therefore not sure what the appendto rules/order is.

The good news is that it works, but I am a little concerned that it _just happens_ to work and will one day break 🙃 

I was seeing some other weird behaviour with this as well, such as these sections being discarded despite their labels being called elsewhere. I've just had to add `keep` to them to get around this for now.

vhelin, 2023-09-20:
I'll return to this issue tomorrow, but if my memory serves me right the .SECTIONs are sorted by their PRIORITY and then by their size, biggest come first. Equal size .SECTIONs are processed in the order qsort() gives them so it might be random. :) Then when the .SECTIONs are sorted they are processed in that order in APPENDTO etc. processes. I'd suggest you give .SECTIONs priorities to have them appended in the order you want.

I could be wrong, so more tomorrow when I have the time to check this out.

vhelin, 2023-09-20:
I also think that I should improve the documentation of APPENDTO and AFTER keywords in .SECTIONs after I've checked out how it really works so that my memory is not telling me lies, tomorrow.

vhelin, 2023-09-21:
Yep, what I remembered was correct. The order where the .SECTIONs are sorted initially in linker dictates the order in which they are appended.

1. No priority (or all have default 0 priority) - bigger .SECTIONs are processed (i.e., appended) first

[image]

2. Use PRIORITY to change the processing order of .SECTIONs

[image]

vhelin, 2023-09-21:
> The good news is that it works, but I am a little concerned that it _just happens_ to work and will one day break 🙃

Yes, I think in this case you've been lucky and the sizes of the appended .SECTIONs happen to be so that the size sorting puts them in the order you want. :)

> I was seeing some other weird behaviour with this as well, such as these sections being discarded despite their labels being called elsewhere. I've just had to add `keep` to them to get around this for now. It seems to do this for non-empty sections as well like [this one](https://github.com/lajohnston/zest/blob/09139100870c3be112cdaebf37419e6f92d125dd/src/postTest.asm#L4).

About this, I'm looking at your project and not seeing a way to compile it so I don't know how to run it and see what WLA DX does there... Any way you could produce perhaps a small example project where this happens?

lajohnston, 2023-09-21:
> About this, I'm looking at your project and not seeing a way to compile it so I don't know how to run it and see what WLA DX does there... Any way you could produce perhaps a small example project where this happens? Or just let me know how to run that project through WLA DX...

Thanks, there's a [build.sh](https://github.com/lajohnston/zest/blob/main/examples/build.sh) script that builds each directory in the examples project. The one that demonstrates it is the [09-hooks](https://github.com/lajohnston/zest/tree/main/examples/09-hooks) one, so could just build the suite.asm file for that and it should pull in the files it needs.

I've taken the `keep`s out in the [preSuite](https://github.com/lajohnston/zest/blob/29828f2369f38b87f7ad479e2ebb6d3024b985b1/src/preSuite.asm), [preTest](https://github.com/lajohnston/zest/blob/29828f2369f38b87f7ad479e2ebb6d3024b985b1/src/preTest.asm) and [postTest](https://github.com/lajohnston/zest/blob/5a781c170dc9de41b769ea91db4588c747bf9502/src/postTest.asm) files and it's still working, so perhaps it's actually fine. However, when passing `-v` to `wlalink` it says it's discarding the sections like `../.././src/preTest.asm: Section "zest.preTest.end" was discarded`, when it actually isn't, so perhaps that's what was happening

vhelin, 2023-09-21:
> Thanks, there's a [build.sh](https://github.com/lajohnston/zest/blob/main/examples/build.sh) script that builds each directory in the examples project. The one that demonstrates it is the [...]

Ah, it was in the examples folder! I'll take a look at it...

> I've taken the `keep`s out in the [preSuite](https://github.com/lajohnston/zest/blob/29828f2369f38b87f7ad479e2ebb6d3024b985b1/src/preSuite.asm), [...]

Hmm let me investigate

vhelin, 2023-09-21:
I think I'm blind as I'm not seeing "Section "zest.preTest.end" was discarded" in WLALINK's output in any of the examples... :)

vhelin, 2023-09-21:
Maybe you removing "KEEP" makes those .SECTIONs being discarded? Or WLALINK thinks it's doing that but itsn't...

vhelin, 2023-09-21:
Ok, removed KEEP from src/preSuite.asm but now I get this

[image]

I'm thinking that removing KEEP from zest.PreSuite .SECTION make WLALINK to drop it as you are giving -d to WLALINK (to discard empty .SECTIONs).

vhelin, 2023-09-21:
Oh, LOL, I added -d for WLALINK to my test case and it also shows it's discarding .SECTIONs that were using APPENDTO. I guesstimate the .SECTION appender appends the .SECTIONs, but then turns them to zero size after that and then they are discarded for being empty. I'll investigate more...

vhelin, 2023-09-21:
Yeah, that was the case - WLALINK was saying it was discarding a .SECTION that had been APPENDTO another .SECTION - the discarded fprintf() for that .SECTION was incorrect, but it should have worked just fine. Just add PRORITY to your .SECTIONs that your are APPENDTO and they should be in correct order.

lajohnston, 2023-09-22:
Thanks, that saves some confusion :)

I think I've worked this out now. I hadn't realised, but it seems PRIORITY accepts negative values, thus making these sections lower priority than the default (0) and so always positioned last even if another appended section forgets to specify a priority. This is exactly what I want for the footer so seems like the solution (unless there are any issues with this you can think of). If this is intended I don't know whether it's worth mentioning it in the docs if it isn't already. -9999999999  appears to be the lowest it can go.

So in summary:
1. `AFTER` works a bit like fixed sections (section + offset), so can't be allocated flexibly and can cause allocation clashes
2. `APPENDTO` allows more flexible allocation. The append order can be altered with `PRIORITY` (the higher the number, the higher in the list it goes). With a negative priority a section can be appended to the end of a sequence of appended sections like a footer

The doc improvements and the DISCARD warning fix seem good so I'm happy to close this issue. Thanks for your help :)

closed 2023-09-22 by lajohnston