vhelin/wla-dx #613: [MS-DOS] Compiling under MS-DOS giving me a missing source file. TGB.C I need TGB.o

Kitsune64, 2023-11-06:
I attempt to make **W-GB.EXE** through DJGPP (GCC compiler for MS-DOS)
At the end of compiling ***.o** files, it give me this error:
No rules to make target 'tgb.c', needed by 'tgb.o'. Stop

I don't find this file in the folders of wla-dx 10.5

vhelin, 2023-11-06:
I have to admit that I have no part in the MS-DOS build scripts, but that tgb.c is something that is generated compiling and running instruction_table_generator (with GB=1 defined?).

[tgb.txt](https://github.com/vhelin/wla-dx/files/13272412/tgb.txt)

If you are using the latest sources you can use this (rename it to tgb.c), but yeah, someone with knowledge on the MS-DOS side here could hopefully fix the build script(s)? :)

Kitsune64, 2023-11-06:
Oh that worked directly with this little file. Thank you!

I have manually  compiled some files as example for obtain WLAD.EXE and WLAB.EXE
I have in start to read in makefiles too
Here an example:
`GCC -ansi -O3 -pedantic -Wall -DMSDOS -DGB main.c -o wlad.exe`
This is the line for compiling wlad.exe.

I don't remind all I have does...But in finish the compilating work with some research and the file you give in this post.

I have copy the file "./historical/makefiles/makefile.msdos.gb" for DOS (Gameboy console interest me) in the main(root) directory of WLA-DX manually launch every makefile for msdos I find for the GameBoy cause it the console I target for now.

I don't really know how much executable I must have in the end and I'm not mastering Makefiles to create a proper script.

At the begining I find the **msdos.bat file**, it's a good idee to use BAT file for dos but I had some trouble to compiling with this one, it not really working BUT this same BAT file help me to understand what to do and where to find the file I must compile as same as the Makefiles helped me to understand how use GCC (it give me the good parameters to compile file). 

So I can't create a script for now...But I propose to publish the binary for MS-DOS when I have finish to compile the executables files.

For now I have:

1. **WLAB.EXE** v1.2
2. **WLAD.EXE** v1.3
3. **WLALINK.EXE** (but I can't view the version when I launch it the list of command take all the screen of DOS)
4. **W-GB.EXE** (Same as above, I can't view the version...)

It's all files I have compiled for now and I don't know how much executable I must have in final.

I will continue to read the **msdos.bat** file cause it list all the **makefiles** to launch for compiling...Don't forget to put this tgb.txt as a *.C in your future releases. 

Thanks!

vhelin, 2023-11-06:
> Don't forget to put this tgb.txt as a *.C in your future releases.

Actually the point in having those files generated as you compile the assembler is that whenever there are chnages in the instruction arrays the optimization arrays in those files are regenerated or else they'd assembler would not just function. Thus I'm not adding them to GitHub at least, but it might be nice to add them to releases, hmm....

[ins_tbl_gen.zip](https://github.com/vhelin/wla-dx/files/13272884/ins_tbl_gen.zip)

Here are all the generated files for v10.6 which I'm hoping to release later this week after I get new issues in the Amiga build fixed.

Looking at our makefile generator, it's in historical/makefile_generator/gen.py - it seems to generate MSDOS makefiles as well so if the MSDOS makefiles are not working then fixing gen.py would be really useful in the long run!

It would be great to have MSDOS executables for releases and if we could do those almost automatically - that'd be great! :)

@lhsazevedo you did the makefile generator script if I remember correctly?

vhelin, 2023-11-06:
Oh nice, you got it to work!? Good news! :)

Kitsune64, 2023-11-06:
thanks, it ask me about these file for the next completion. **tsuperfx.c** I never worked with WLA-DX for create a game, then I can't say if it work at 100%, but the compilated EXE run in DOS environment here.

I will try to extract the EXE I have on the Virtual Machine and send it here as a join file in the message you will can try to use these with Dosbox maybe.

vhelin, 2023-11-06:
I'm not actually using WLA DX for any personal projects at the moment so no need to do those builds for me, but perhaps the users here would appreciate them?

I'm also thinking that I could try to add as a final feature to v10.6 a pause for those WLA & WLALINK help screens to the MSDOS build. Is the MSDOS screen like 80x25 characters? I personally never used MSDOS...

Kitsune64, 2023-11-06:
I never count the number of characters on screen ^^ but it's seem exact when I search on web 80x25.
I have attached the compiled files I have done for now.

Here is the file for DOS (or Dosbox):
[WLADXDOS.ZIP](https://github.com/vhelin/wla-dx/files/13273181/WLADXDOS.ZIP)

vhelin, 2023-11-06:
Thank you for the binaries, good work! :)

Just went through the options of my Cygwin window and it seems I can set it presicely to 80x25 characters!

[image]

... so the pausing is definitely coming to v10.6 if I can do it with getc() or fgets() etc. At least for the info screen. Later perhaps make the assembly/results time prints also support it?

vhelin, 2023-11-06:
I have to admit that I don't know what's the default defacto behavior of an app in MSDOS that displays more than one screen worth of text. I'm guess it's pause. Please correct me people if I'm wrong.

Kitsune64, 2023-11-06:
That depend on the shoulders of the developper ;)

If you don't make a pause in the developped software and show a lot of text the screen scroll down automatically.
As example the **DIR** command in MS-DOS (a basic command for listing directories) if there are a lot of folders to list we can't see all these BUT if I use **DIR /P** command...It will show Directories list as different "pages/screen". 
And in MS-DOS you can't scroll up like a Linux Terminal on a Desktop environment. I don't know how DOS alternative work about it (for FreeDOS I think)

vhelin, 2023-11-06:
Ok so a separate option for pausing? I don't think there's a way in pure ANSI C to know the current "screen" resolution as there is no concept of screen / window so we might need additional libraries for that for MSDOS builds. Maybe? Maybe something to think about for future releases.

If we wanted to go platform independent with this feature we could let the user specify the screen size:

wla-gb -SX 80 -SY 25 -p -o main.o main.s

Kitsune64, 2023-11-06:
80*25 is a standard use of mode TEXT for MS-DOS you can use it of course.

"By far the most common text mode used in DOS environments, and initial Windows consoles, is the default 80 columns by 25 rows, or 80×25, with 16 colors. This mode was available on practically all [IBM](https://en.wikipedia.org/wiki/IBM) and compatible personal computers. Several programs, such as [terminal emulators](https://en.wikipedia.org/wiki/Terminal_emulator), used only 80×24 for the main display and reserved the bottom row for a [status bar](https://en.wikipedia.org/wiki/Status_bar)."

Wikipedia Source:
[https://en.wikipedia.org/wiki/Text_mode]

vhelin, 2023-11-06:
Ah, ok, 80x24 for WLA/WLALINK text and then last line would say something like "press a key to continue..."

I made an issue/feature request about this: https://github.com/vhelin/wla-dx/issues/614

Cannot do all that for v10.6, but at least I can get the task started with the info texts...

Kitsune64, 2023-11-07:
I tested the compilation for Gameboy on Dosbox (DOS emulator) with the compiled WLA-DX executables above and it worked.

Tools used from WLA-DX for compiling a GameBoy game:
**W-GB**
**WLALINK**

Based on this source-code at this link:
[gameboy template for WLA-DX](https://gist.github.com/iamgreaser/9cccd24dd4519a0ae95240d8b8eeb252)

I make a custom **build.bat** file for compiling in a child folder:
```
..\W-GB -o MAIN.O MAIN.ASM
..\WLALINK -r LINKFILE MAIN.GB
```
**main.asm** and **linkfile** files must be in a child folder and the root folder of WLA-DX must contain **W-GB.EXE** and **WLALINK.EXE** for this example.

The result is attach to this message
[image]

I'm near to finish to compile all the WLA-DX 10.5 pack I think. I will approch the 18 binaries files listed in msdos.bat file from your release. All of of these executables have been tested to run on MS-DOS 6.22. I will attach files as an archive here when I will finish to compile this release.

lhsazevedo, 2023-11-07:
> @lhsazevedo you did the makefile generator script if I remember correctly?

Yup, that was me! Anything we need to fix in it?

vhelin, 2023-11-07:
If I understand @Kitsune64 here correctly he has two problems due to MSDOS's 8.3 filename restrictions

1. The msdos.bat doesn't run instruction_table_generator for all architectures so that we'd get the code generated t*.c files. I'm not sure you worked on msdot.bat, though...
2. The generated makefiles are not as well obeying the 8.3 name restrictions. Maybe have msdos folder (and amiga folder for amiga files) under makefiles where 8.3 msdos makefiles would go?

@Kitsune64 am I correct? You have other improvement ideas?

Kitsune64, 2023-11-07:
Answers:
 1.Effectively the MSDOS.BAT file don't launch the Makefile about Instruction table generator and I couldn't find the makefile to generate the instruction table but if it exist in the release maybe this makefile is functional? It's just that I didn't see it.

2.Yes, the restricting file names was an issue under MSDOS,

But I definitely missed some steps when I worked around. I did some steps manually like copy+pasting makefiles for MSDOS into the main WLA-DX directory. I also ran each makefile manually with these commands:
make -f makefile.dos (for compiling)
make -f makefile.dos clean (for cleaning project)
the *.dos is an example here. There are not *.dos file in the WLA-DX release but it can be the way to go (it respect the limit of 3 letters for extention)

Details:
I worked with MS-DOS 6.22, the DEL command in end of Makefiles for DOS use the parameter /F /Q which does not work with the version I used (I edited each Makefile to remove these "/F / Q" to make it work)
It depends on which MS-DOS version you are targeting.

I was wondering why not make a BAT file that would automate the WHOLE compilation procedure? So copying the files to the right place and launching the Makefile would be part of the automatic procedure. Looks like that's what it was supposed to do but the BAT file failed to compile the programs here.

vhelin, 2023-11-10:
> I'm near to finish to compile all the WLA-DX 10.5 pack I think. I will approch the 18 binaries files listed in msdos.bat file from your release. All of of these executables have been tested to run on MS-DOS 6.22. I will attach files as an archive here when I will finish to compile the release.
> 
> It's done! [WLADXDOS.ZIP](https://github.com/vhelin/wla-dx/files/13277919/WLADXDOS.ZIP)

Thank you for these! I'll add them to the v10.5 release here on GitHub this weekend!

vhelin, 2023-11-12:
Good news! If you take the latest sources both the assembler and the linker support -p for pausing and -SX (default 80) and -SY (default 24) for screen/console size definition. It wasn't actually such a big thing to implement fully so we'll have this in v10.6 I'll release hopefully in one week.

vhelin, 2023-11-12:
> I was wondering why not make a BAT file that would automate the WHOLE compilation procedure? So copying the files to the right place and launching the Makefile would be part of the automatic [...]

I have to admit that you are the first person I've ever used to use WLA DX under MSDOS. As I've never really used MSDOS in my life in the past I don't know how to make such a .BAT file, right off the bat. :) I could learn, though, but don't see that as a high priority task on my TODO list.

I'm guessing "instruction_table_generator" folder name is too long for a .BAT file, but aren't there MSDOS'es that support longer than 8.3 file names? Then again renaming that folder to fit 8.3 is not a big deal... But then again are the executables compiled under such a version of MSDOS compatible with MSDOS 6.22, and is 6.22 the version we should be targeting? I know nothing of the MSDOS world...