vhelin/wla-dx #614: Output all texts through a function that pauses every time when we've outputted one screen worth of text

vhelin, 2023-11-06:
The origin of this feature is MSDOS. I think it's a 80x25 characters per screen, but we could add support for other screen sizes with little cost:

wla-gb -SX 80 -SY 25 -p -o main.o main.s

Note that lines wider than 80 characters in this example will be counted as 2+ lines...

vhelin, 2023-11-06:
80x24 in this case for WLA/WLALINK's text and last line could show something like "Press any key to continue..."

I'm hoping it's doable using getc() or fgets() or something similar.

vhelin, 2023-11-12:
Tomorrow I'll add this feature to the linker...

cr1901, 2023-11-12:
FWIW, even early DOSes have the `MORE` command:

```
TYPE FILE.TXT | MORE
```

Yes, redirection works in DOS lol (it's implemented by writing to a temporary file).

vhelin, 2023-11-12:
Oh man. :) Well, I'll add this feature anyway, perhaps there are platforms where it would be useful? Like on the Amiga, but there you could manually stop/resume the printing or install something like KingCon...

closed 2023-11-12 by vhelin