vhelin/wla-dx #617: Nested macro loses argument type

willbritton, 2023-12-10:
Following documentation [here](https://wla-dx.readthedocs.io/en/latest/asmdiv.html#macro-test) as well as discussion [here](https://github.com/vhelin/wla-dx/issues/69) I established that using `\?1` etc. is the only way to write a macro for 6502 which would work equally well for immediate mode as well as zero page mode arguments.

This appears to work fine for a single simple macro (`LOAD_DATA` in the example below) but when calling a macro from within another macro it seems like the argument type gets lost (`COPY_DATA` with immediate arg in the example below).

(Side note: I still haven't managed to get my head around why the immediacy of the argument does get removed at all - I understand macros aren't simple token replacements in wla but it would feel like the intuitive behaviour of macros in situations like this is to let the assembler make the decisions about addressing mode.)

```
.memorymap
  defaultslot 0
  slotsize $100
  slot 0 $0000
.endme
.rombankmap
  bankstotal 1
  banksize $100
  banks 1
.endro

.macro LOAD_DATA
  .if \?1 == ARG_IMMEDIATE
    lda #\1
  .else
    lda \1
  .endif
.endm

.macro COPY_DATA
  LOAD_DATA(\1)
  sta \2
.endm

.bank 0 slot 0
.orga $0000

LOAD_DATA($aa)       ; emits      A5 AA (lda $aa)  - correct
LOAD_DATA(#$aa)      ; emits      A9 AA (lda #$aa) - correct
COPY_DATA($55, $00)  ; emits      A5 55 (lda $55) 85 00 (sta $00) - correct
COPY_DATA(#$55, $00) ; emits      A5 55 (lda $55) 85 00 (sta $00) - incorrect
```

vhelin, 2023-12-10:
I assume we are not copying the argument type when copying the arguments to a new .MACRO call. I'll try to fix this asap!

vhelin, 2025-09-15:
If this issue is still valid, please reopen this issue. AFAIK my commit fixed this issue back in the day...