vhelin/wla-dx #618: [documentation] _sizeof generated inside .section even without following label

willbritton, 2023-12-10:
This is a minor observation I made accidentally, but potentially a little bit of clarification could be added somewhere in the docs about how `_sizeof` symbols are generated.

Here's what I think:

1. Outside of `.section` blocks, `_sizeof` symbols are generated between one non-child label and another non-child label, as described [here](https://wla-dx.readthedocs.io/en/latest/asmsyntax.html#labels).

e.g. 
```
.my_first_label:
  .db $55
.my_second_label:
  .db $aa
```

In this case, a symbol `_sizeof_my_first_label` is generated with value 1.

2. Outside of `.section` blocks, no `_sizeof` symbol is generated for the very last label in the source, e.g.:
```
.my_last_label:
  .db $33
; end of file
```

A symbol `_sizeof_my_last_label` is NOT generated in this case.

3. Inside a `.section` block, a `_sizeof` symbol is generated as in case 1 above between two labels:
```
.section "MySection" bank 0 slot 0 free
.my_first_label:
  .db $55
.my_second_label:
  .db $aa
.ends
```

In this case, a symbol `_sizeof_my_first_label` is generated with value 1.

4. Inside a `.section` block, a `_sizeof` symbol is generated between the last label and the end of the block, unlike in case 2 above:
```
.section "MySection" bank 0 slot 0 free
.my_last_label:
  .db $33
.ends
```

In this case a symbol `_sizeof_my_last_label` IS generated.

Hope I have this correct! Assuming it's intentional, perhaps the docs should reflect it more clearly.

maxim-zhao, 2023-12-27:
I seem to remember this was added at my request, as else I had to add labels to make it work. I concur that the end of the assigned data ought to act the same way, so for example it wouldn't include any unassigned (empty) bytes in the size of that last label, or if any "gaps" are left due to use of .org to move the insertion point around.

I guess in this case:
```
.org 0
label1: .db "Hello world"

.org $100
label2: .db 0
```
`_sizeof_label1` ought to be 11; is is actually 256 now?

vhelin, 2024-01-21:
Ok, looking at write.c:generate_sizeof_label_definitions() in wlalink I can say that the _sizeof_* calculator only uses labels, but by rewriting that function I could make it look into the ROM usage array to see the used and unused bytes of the output. Not 100% sure at this point if the final information is available at the time we generate _sizeof_* labels, but I'll dig into this next week.

vhelin, 2024-02-05:
I hope the latest commit fixes this issue: I made generate_sizeof_label_definitions() in WLALINK to stop when it encounters gaps in the output outside .SECTIONs (.SECTIONs don't have gaps inside them). I can still see that the function calculates _sizeof_* labels for definitions, which I guess it should not, but I'll fix that next.

Also sorry for my absense, I've been doing other things, taking a break from WLA DX, but I'll try to devote more time to WLA DX in the near future...

vhelin, 2024-02-05:
My latest commit has a bug it seems, but I'll try to fix it tomorrow...

vhelin, 2024-02-05:
Ok, if you download the latest sources the problems mentioned in this issue should be gone. I also updated the documentation to tell what happens with labels outside .SECTIONs (gaps in output that follow the labels)