vhelin/wla-dx #621: [Not an issue] Where to seek for help ?

cgimenez, 2024-01-29:
Hi all, and thanks for wla author(s), having an assembler for a such range of CPU is amazing !

This issue is not an issue but rather a question to how to get help with wla ? Is there any forum or community space ?
I've not yet assembled anything but I've read all the assembler directives and there still some question : 

if foo is declared as
```
foo:
.DB 1, 0
```

How can I get the hi and low bytes of its adress ?
I presume it might be something like
`LDA #>foo`
and
`LDA #<foo`

But I did not managed to find any clue in the doc, or I missed it badly ;-)

Neui, 2024-01-29:
GitHub provides the [Discussions tab](https://github.com/vhelin/wla-dx/discussions) which this project uses.

cgimenez, 2024-01-29:
I totally missed the github discussions ;-)

Thanks !

closed 2024-01-29 by cgimenez