vhelin/wla-dx #624: Generating Symbol File Information on Library Crashes WLA-6502

ToriningenGames, 2024-03-29:
WLA-6502 Version: Commit [4605618](https://github.com/vhelin/wla-dx/commit/4605618716566584254025ed44d3c20995a07776)
Host Info: MX Linux 21.3 (Debian 11.9) x86_64, glibc 2.31-13+deb11u8, GCC 10.2.1
Reproduction:
Run `wla-6502 -i -l Test.o Test.txt` with [this file](https://github.com/vhelin/wla-dx/files/14797278/Test.txt).
Expected result: Successful Assembly
Actual result: Crashes with `malloc(): invalid size (unsorted)`

Quick analysis with valgrind says that `listfile.c` writes past the end of the `g_global_listfile_ints` and `g_global_listfile_cmds` arrays when processing a `k` command, suggesting `g_global_listfile_items` is too short for some reason. The actual crash is at the next call to calloc.
I think it happens regardless of the provided file, and may happen with every WLA assembler, but I didn't thoroughly check.

vhelin, 2024-03-29:
Thanks for the report! My four day Easter holiday just started and I hope I can fix this during those days. :)

vhelin, 2024-04-01:
It seems that listfile support for library files was quite broken. I think I need to add a test for them...

ToriningenGames, 2024-04-06:
It works in my case now, but I didn't realize that for a bit because the list files were generated in my source directory. Is this intended behavior? Personally, I'd like them to go with the objects, aka output folder vs. input folder. This keeps things cleaner and enables out-of-tree builds of projects, something I'm rather fond of.

vhelin, 2024-04-06:
That's the intended behavior, but what you suggest is a great feature idea, maybe put that option behind some flag. I'll make a separate issue about it.

closed 2024-04-06 by vhelin