vhelin/wla-dx #630: Issue with fewer banks than slots

lajohnston, 2024-04-08:
Hi there,

Sorry, I'm seeing a strange mapping issue whereby the following code results in the `bank1Code` and `bank2Code` labels being assigned to the same address despite being in different banks:

```asm
.memorymap
    defaultslot 1
    slotsize $4000

    slot 0 $0000  ; ROM
    slot 1 $4000  ; ROM
    slot 2 $8000  ; ROM
    slot 3 $c000  ; RAM
.endme

; 3x16 KB ROM Banks
.rombankmap
    bankstotal 3      
    banksize $4000
    banks 3            
.endro

.section "bank1Code" free slot 1 bank 1 keep
    bank1Code:
        ld a, 1
        ret
.ends

.section "bank2Code" free slot 1 bank 2 keep
    bank2Code:
        ld a, 2
        ret
.ends
```

The `wla-z80` output indicates they are mapped to the correct banks:
```
ROM bank 1 (3 bytes (0.02%) used)
  - Outside .SECTIONs (0 bytes)
    - No data outside .SECTIONs.
  - Sections (3 bytes)
    - .SECTION "bank1Code" (3 bytes).
ROM bank 2 (3 bytes (0.02%) used)
  - Outside .SECTIONs (0 bytes)
    - No data outside .SECTIONs.
  - Sections (3 bytes)
    - .SECTION "bank2Code" (3 bytes).
```

...but when running in an emulator like Emulicious I can see both labels occupying the bank 1 address. Changing the `rombankmap` to contain 4 banks seems to fix the issue, so is it related to having fewer banks than there are slots?

vhelin, 2024-04-08:
Thanks for this report also, I'll try to fix this asap...

vhelin, 2024-04-08:
Ok, I put your code inside code.s and ran this

  wla-z80 -i -k -v -o code.o code.s

Then I ran (code.o being in linkfile's [objectives]

  wlalink -v -S -i linkfile linked.rom

->

code.lst

[image]

linked.sym

[image]

Used hex editor in Emacs to see that the .SECTIONs are where they should be, so I'm guessing your emulator doesn't work properly with ROMs that have 3 banks. Does the platform (SMS?) support 3 bank ROMs?

vhelin, 2024-04-08:
One thing that makes me wonder is this output from wla-z80

[image]

Where's the ROM info? Need to investigate...

lajohnston, 2024-04-09:
> I'm guessing your emulator doesn't work properly with ROMs that have 3 banks

Ah thanks, yes I've just tried the ROM on a real system and it works fine, so I'll reach out to the emulator maintainer to see if they are able to help :)

lajohnston, 2024-04-09:
Thanks for looking into this. It seems some emulators treat 48KB ROMs as not having a hardware mapper. I've padded my ROM to 64KB to prevent that issue so I'm happy to close this now :)

closed 2024-04-09 by lajohnston

vhelin, 2024-04-10:
I'm glad I could help! :)