vhelin/wla-dx #631: Executeables not actually working on XP with PIII

Kroc, 2024-05-02:
I have a Pentium III sub-notebook I'd like to use to do some CP/M coding. My project [v80](https://github.com/Kroc/v80) uses WLA-DX to bootstrap a Z80 assembler in Z80. Using the 10.6 release the binaries crash with an access violation.

Knowing that the Pentium III doesn't have SSE2 -- I already had to change the VS solution for [RunCPM](https://github.com/MockbaTheBorg/RunCPM) -- I changed the v10.6-tag 32-bit release build to use NO vector extensions, not even SSE1 and recompiled but even this doesn't work.

[image]

Do you have any means to test a real WindowsXP machine, maybe one without SSE2?  
I'll try get you some more details once I'm back home with the Pentium III laptop.

vhelin, 2024-05-02:
Unfortunately I don't have any such old computers with Windows XP. :( Even my parents' retired old computer is too new... I think the SDK we use to build WLA DX should allow the executables to run under Windows XP, but yeah, haven't tested that. I guess, like you say, now it's up to get the settings correct for old CPUs...

If you can do that, please let us have the modified Visual Studio solution!

Kroc, 2024-05-02:
I'll tinker with it and see what I can do!

Kroc, 2024-05-02:
Okay, I got it working, I didn't realise I was applying changes to only one project and not every project -- Visual Studio's solution management is pure awful.

[image]

vhelin, 2024-05-02:
Oh yeah, please let us know what you did and make a pull request out of the solution file if you think it's something that would work for everybody. I don't think we need SSE2 etc. flags in master... :)

vhelin, 2024-05-07:
Well, I did the same to x86/Win32 in the Visual Studio projects

[image]

I hope that's enough. I assume we can close this issue.

closed 2024-05-07 by vhelin

Kroc, 2024-05-07:
I've been very busy and not at the place where my PIII laptop is but I will test once I get there.
I would recommend changing compiler optimisation to favour size over speed for x86 so that the code doesn't blow out the cache on old CPUs making them run much slower. A crappy modern CPU might have 128KB of L1 cache but that's the entire L2 cache of a PIII and its L1 is only 8KB!

vhelin, 2024-05-07:
Done!